WildernessLabs/Meadow.Foundation #6: Bug with PwmLed on port D13

jorgedevs, 2019-09-06:
When turning on port D12, por D13 also turns on.

```
using Meadow;
using Meadow.Devices;
using Meadow.Foundation.Leds;
using System;
using System.Collections.Generic;
using System.Threading;

namespace PwmLed_Sample
{
    public class PwmLedApp : App<F7Micro, PwmLedApp>
    {
        List<PwmLed> pwmLeds;

        public PwmLedApp()
        {
            Console.WriteLine("Initializing...");

            pwmLeds = new List<PwmLed>();
            pwmLeds.Add(new PwmLed(Device.CreatePwmPort(Device.Pins.D12), TypicalForwardVoltage.Green));
            pwmLeds.Add(new PwmLed(Device.CreatePwmPort(Device.Pins.D13), TypicalForwardVoltage.Blue));

            TestPwmLeds();
        }

        protected void TestPwmLeds()
        {
            Console.WriteLine("TestPwmLeds...");

            while (true)
            {
                Console.WriteLine("Turning on and off each led for 1 second");
                foreach (var pwmLed in pwmLeds)
                {
                    pwmLed.IsOn = true;
                    Thread.Sleep(500);
                    pwmLed.IsOn = false;
                }
            }
        }
    }
}
```

csuski, 2019-10-19:
I am not able to reproduce this on version 0.10.4

jorgedevs, 2019-10-19:
you're right, this is no longer happening with the latest version.

closed 2019-10-19 by jorgedevs