WildernessLabs/Meadow.Foundation #10: Meadow Freeze w/ timer.

BillAgner, 2019-10-17:
My code is event driven from rotary encoder and button. Then I'm using a timer event (System.Threading.Timer) to check encoder values and update LCD. In this timer event handler, I coded onboard led up for a heartbeat. Once the application is instantiated in the main function, I just Thread.Sleep(Timeout.Infinite); After some time heartbeat stops and rotary encoder rotation value stop being updated on LCD. Also after previous post, I was sure to hit reset before deploying.

The *.txt files are renamed *.cs

[RotaryEncoder.txt](https://github.com/WildernessLabs/Meadow.Foundation/files/3740544/RotaryEncoder.txt)
[MeadowApp.txt](https://github.com/WildernessLabs/Meadow.Foundation/files/3740545/MeadowApp.txt)
[Program.txt](https://github.com/WildernessLabs/Meadow.Foundation/files/3740546/Program.txt)

gismofx, 2019-11-06:
Maybe related to the issue I found as well:
https://github.com/WildernessLabs/Meadow_Issues/issues/28

bryancostanich, 2019-11-22:
I believe this may be a duplicate of this: https://github.com/WildernessLabs/Meadow_Issues/issues/37

adrianstevens, 2020-11-26:
fixed in b4.0 - closing issue

closed 2020-11-26 by adrianstevens