WildernessLabs/Meadow.Foundation #11: Timing issue with stopping an LED from blinking and turning it on

csuski, 2019-10-19:
Running the code below results in the LED staying off from the output 'Turn back on' all the way through the end of the loop.

```
using System;
using System.Threading;
using Meadow;
using Meadow.Foundation.Leds;
using Meadow.Devices;

namespace LedSample
{
    public class Program
    {
        static IApp _app;
        public static void Main()
        {
            _app = new LedApp();
        }
    }

    public class LedApp : App<F7Micro, LedApp>
    {
        public LedApp()
        {
            // create a new Led on pin 8
            var led = new Led(Device.CreateDigitalOutputPort(Device.Pins.D08));

            while (true)
            {
                Console.WriteLine("Start On");
                led.IsOn = true;    // Led ON
                Thread.Sleep(3000); // 3 seconds
                Console.WriteLine("Start Blink");
                led.StartBlink(2000, 2000);
                Thread.Sleep(1000); // 5 seconds
                Console.WriteLine("Stop Blink");
                led.Stop();
                Console.WriteLine("Turn back on");
                led.IsOn = true;
                Thread.Sleep(5000);
                Console.WriteLine("End Loop");
            }
        }
    }
}
```

This is because the LED is set to on, while it is currently in the On step of blink and then it gets set to off in the blink task when it finishes the blink.

There are a lot of different solutions here, but I think my assumption using the code is that once I call led.Stop(), the blink command will not change state again and it will stop in whatever state it currently is. 

This could be accomplished by checking the running state before turning the LED off and if it is no longer running to break the while loop.

gasia89, 2019-11-10:
I can take this one as well.

gasia89, 2019-11-10:
This is actually a problem with the setter on IsOn. It looks like this was made more complicated than it needs to be by implimenting a protected field, but in reality, the port is already public to get, and protected to set, so why bother? I feel it should be simplified to:
`
        public bool IsOn
        {
            get { return Port.State; }
            set { Port.State = value; }
        }
`

Then when your code stops the animation, the `Stop()` method turns off the led, as it is supposed to, and your `IsOn = true` works as expected.

gasia89, 2019-11-10:
https://github.com/WildernessLabs/Meadow.Foundation/pull/21

bryancostanich, 2019-11-22:
See pull request.

jorgedevs, 2019-11-22:
PR reviewed and merged. Thanks!

closed 2019-11-22 by jorgedevs