WildernessLabs/Meadow.Foundation #12: Calling Led.StartBlink when the Led is on causes a flicker

csuski, 2019-10-19:
If you call Led.StartBlink() on a Led that is currently set to on it turns it off briefly before turning it back on at the beginning of the blink cycle.

I found this unexpected and the only way to figure out why it was flickering was to look at the implementation of StartBlink().  

I think turning off the LED before the start of the blink sequence should be removed. It forces anyone who wants to go from the ON state to the Blink state to have a flicker and there is no way around it without writing your own implementation (and I can't imagine this is the desired before for anyone).

gasia89, 2019-11-10:
I have a fix for this, but I am having permissions issues with the repo at the moment.

gasia89, 2019-11-10:
Pull request [22](https://github.com/WildernessLabs/Meadow.Foundation/pull/22) fixes this.

bryancostanich, 2019-11-22:
See above mentioned pull request.

jorgedevs, 2019-11-22:
PR reviewed and merged. Thanks!

closed 2019-11-22 by jorgedevs