WildernessLabs/Meadow.Foundation #13: PushButton.State does not sync with Started/Ended events

csuski, 2019-10-19:
If you output the PushButton.State it will always output True for low and False for high.

However, if you look at the PushButton.PressStarted and PushButton.PressEnded events, PressStarted fires going from Low to High and PressEnded first from High to Low.

If you wire the button as a Pull Up or Pull Down something will always be wrong.

Sample code to demonstrate

```
using System.Threading;
using Meadow;
using Meadow.Devices;
using Meadow.Foundation.Sensors.Buttons;
using System;

namespace PushButtonSample
{
    public class Program
    {
        static IApp _app;
        public static void Main()
        {
            _app = new PushButtonApp();
        }
    }

    public class PushButtonApp : App<F7Micro, PushButtonApp>
    {
        public PushButtonApp()
        {
            var pushButton = new PushButton(Device, Device.Pins.D08,40);
            pushButton.PressStarted += PushButtonPressStarted;
            pushButton.PressEnded += PushButtonPressEnded;

            while (true)
            {
                Console.WriteLine($"State: {pushButton.State}");
                Thread.Sleep(2000);
            }
        }

        private void PushButtonPressStarted(object sender, EventArgs e)
        {
            Console.WriteLine("Press Started...");
        }

        private void PushButtonPressEnded(object sender, EventArgs e)
        {
            Console.WriteLine("Press Ended...");
        }
    }
}
```

gasia89, 2019-11-10:
@csuski For clarification, are you saying that depending on if you use a pull up or pull down resistor on the button, it swaps the detection of events? ie. firing PushButtonPressEnded when it was actually stating?

csuski, 2019-11-10:
I think the main issue is the PushButton class didn't finish getting implemented. The last checkin has a note that it was not implemented and that was removed, it appears, prematurely.

I can't remember the exact direction right now, but basically if you wire it up as a pull up the state is correct, but he PushButtonPressEnded fires when it starts and if you wire it up as a pull down the events fire correctly, but the state is wrong.

If you look at the code it appears there there was the beginning of implementing the CircuitTerminationType property, however, it is only implemented for the button presses, not the state. Also it is protected and not passed in through the constructor so the user can never change it from the default.

gasia89, 2019-11-11:
Wonderful... I will dig in a bit and see what I can do. Still getting my feet wet with Meadow, as mine only showed up a week or so ago and I just opened it up this week.

gasia89, 2019-11-11:
I see what you are referring to. That was probably meant to indicate which type of method we are using to terminate the button circuit, ie. pull up/down. At least thats how I read that. 

I could change the implimentation to swap the event logic based on that value to make that more consistant, but the user would need to specify in the code that it was pull up/pull down. Thoughts?

[image]

gasia89, 2019-11-11:
Ok, looks like there was a resistorMode variable that was being set inside the constructor. Not really sure why. I moved it to the constructor parameters and set it to a default value of disabled to prevent conflicts with others. I put in a pull request to fix this. Here is my suggested new implimentation: 
[image]

csuski, 2019-11-11:
Do the events work properly after that change with pull up and pull down?  The _circuitType field is never set and probably needs to be set based on the resistor mode. So I wouldn't expect the events to output correctly. 

Maybe it could just use the resistor mode and get rid of the circuit type, that way it will work with the other constructor also. 

Also, your formatting is odd. In your screen shot it is correct, but if you look at all your diffs for the pull requests the alignment is off (tabs vs. space?).

This fix might be the fix for this issue as well -> https://github.com/WildernessLabs/Meadow_Issues/issues/18

gasia89, 2019-11-11:
def tabs vs spaces. Space people are so silly. Will take some time to adapt to. 

I see what you mean, cant believe I missed that. Yeah, it seems odd that they are virtually the same.

gasia89, 2019-11-28:
Sorry for the delay, had to order more parts, and get a better testing area setup. Going to be looking at this over the holiday weekend, real-life permitting.

gasia89, 2019-12-02:
Ok, finally got this fixed. See [Pull Request 34](https://github.com/WildernessLabs/Meadow.Foundation/pull/34)

also, in my case, I wired up a pull up and pull down setup side by side. So here is my sample code for reference @csuski : 
`public class MeadowApp : App<F7Micro, MeadowApp>
	{
		public MeadowApp()
		{
			var pushButton_PullUp = new PushButton(Device, Device.Pins.D01, ResistorMode.PullUp, 40);
			var pushButton_PullDown = new PushButton(Device, Device.Pins.D02, ResistorMode.PullDown, 40);

			pushButton_PullUp.PressStarted += PushButtonPressStarted;
			pushButton_PullUp.PressEnded += PushButtonPressEnded;

			pushButton_PullDown.PressStarted += PushButtonPressStarted;
			pushButton_PullDown.PressEnded += PushButtonPressEnded;

			while (true)
			{
				Console.WriteLine($"Button_1 State: {pushButton_PullUp.State}");
				Console.WriteLine($"Button_2 State: {pushButton_PullDown.State}");
				Thread.Sleep(2000);
			}
		}

		private void PushButtonPressStarted(object sender, EventArgs e)
		{
			Console.WriteLine("Press Started...");
		}

		private void PushButtonPressEnded(object sender, EventArgs e)
		{
			Console.WriteLine("Press Ended...");
		}
	}`

adrianstevens, 2020-01-04:
Thanks @gasia89 - I'll try and get this merged over the weekend

Cheers!

ChiefInnovator, 2020-07-17:
We are seeing this issue with the latest code just released this week.  Not sure if the suggested fixes are in or if things are still an issue.  We are going to repro it on Monday and send code.

brucecrombie, 2020-10-03:
Yes - still seeing the same behavior with V0.18.0
When the button is wired from digital port through a pushbutton to ground, the proper resistor mode should be pull up.
When the button is pressed the meadow should fire the pushbutten pressed event on the rising edge. When the pushbutton is released the meadow should fire pushbutton released event on the falling edge.. Instead I get the opposite events.

gasia89, 2020-11-04:
I could have sworn I fixed this last year. Is this still an issue? Or is
this a new issue?

adrianstevens, 2020-12-31:
@jorgedevs is this fixed in the latest updates?

jorgedevs, 2020-12-31:
Yeah, this is fixed in the latest release (v4.2).

closed 2020-12-31 by jorgedevs