WildernessLabs/Meadow.Foundation #19: Support System.Device.Gpio

ethos-tim, 2019-11-01:
Can Meadow support System.Device.Gpio?

https://github.com/dotnet/iot/issues/835

bryancostanich, 2019-11-20:
We have no plans to support that API. The approach we've taken with meadow is far different. Let us know if there are any drivers that there are drivers for in Windows IoT and we'll prioritize porting them!

closed 2019-11-20 by bryancostanich

frogcrush, 2020-11-11:
Hey, would you mind expanding a bit more on how meadow's approach is different? I'm just curious as I feel like it would be beneficial for the community if we reduced the fragmentation in the space. For example, nanoFramework appears to be moving to support dotnet/iot. Based on that linked issue, they also seemed to think it wouldn't be a bad idea to align the ecosystem.