WildernessLabs/Meadow.Foundation #27: AnalogTemperature gives incorrect results for TMP36

LordBenjamin, 2019-11-20:
Connected a TMP36 temperature sensor (ground, 3.3v and Analogue 0) as per this [Arduino tutorial](https://learn.adafruit.com/tmp36-temperature-sensor/using-a-temp-sensor) (except using 3.3v instead of 5v).

The implementation from Meadow.Foundation appears to be:

```c#
float millivolts = await AnalogPort.Read(1, _updateInterval);
float num = millivolts - _yIntercept;
Temperature = num * _millivoltsPerDegreeCentigrade;
```

But doing this (by hand or by using `AnalogTemperature`) gives me very wrong results (e.g. -4,970 °C at room temperature).

By contrast, the following code gives me results around the 20°C mark:

```c#
float yIntercept = 500;
float millivoltsPerDegreeC = 10;

float millivolts = await analogPort.Read(1) * 1000; // results seem to be in V rather than mV
float temp = (millivolts - yIntercept) / millivoltsPerDegreeC; // Need to divide by mV/°C rather than multiplying

Console.WriteLine("{0:N2} millivolts, {1:N2}°C", millivolts, temp);
```
This also matches the maths in the linked Arduino tutorial better than `AnalogTemperature` does.

Am I missing something?

bryancostanich, 2019-11-20:
Looks like a bug! I'll try to update the sensor in the next few days. I've only used the TMP35 and it was working fine.

LordBenjamin, 2019-11-21:
Thanks! Happy to test on the TMP36 once it's been updated.

closed 2020-01-04 by adrianstevens

brucecrombie, 2020-03-22:
TMP35 has same issue as TMP36 - see open issue 46. The correction is in VoltageToTemperature:

var yAdjusted = voltage * 1000 - _yIntercept;
return yAdjusted * _millivoltsPerDegreeCentigrade * 10;

should be

var yAdjusted = voltage * 1000
return (yAdjusted - _yIntercept)  / _millivoltsPerDegreeCentigrade;

commit 55bf08f9 (2020-01-04): Fix temp calculation in AnalogTemperatureSensor - fixes #27