WildernessLabs/Meadow.Foundation #28: NeoPixel Support

atrueresistance, 2019-11-21:
I wired up some [breadboard NeoPixels](https://www.adafruit.com/product/1312) and attempted some sample code. Does anyone know what I'm doing wrong or if NeoPixels are even supported? I can't get even one to blink. 

```
using System;
using System.Threading;
using Meadow;
using Meadow.Devices;
using Meadow.Foundation.Leds;

namespace MeadowApplication1
{
    public class MeadowApp : App<F7Micro, MeadowApp>
    {
        PwmLed neoPixel;

        public MeadowApp()
        {
            ConfigurePorts();
            BlinkLeds();
        }

        public void ConfigurePorts()
        {
            Console.WriteLine("Creating Outputs...");
            neoPixel = new PwmLed(Device.CreatePwmPort(Device.Pins.D02,800), TypicalForwardVoltage.Red);   
        }

        public void BlinkLeds()
        {
            Console.WriteLine("Starting main loop...");
            while (true)
            {
                int wait = 200;

                neoPixel.IsOn = true;
                Thread.Sleep(wait);
                neoPixel.IsOn = false;

            }
        }
    }
}
```
[image]

bryancostanich, 2019-11-21:
I believe we now support the timings necessary in our SPI config to make this work. To be investigated.

bryancostanich, 2019-11-21:
Looks like we still need hold times. From Justin Wilson (@IngenuityMicro), this is the NetMF code that he used to support NeoPixel on Netduino:

```csharp
var spiConfig = new SPI.Configuration(Cpu.Pin.GPIO_NONE, false, 500, 500, false, false, 3200, SPI.SPI_module.SPI1);
            _spi = new SPI(spiConfig);
```

So csSetup and csHoldTime of 500 uSecs and ClockIdleState and ClockEdge of false

gasia89, 2019-12-04:
I have a few spools of neopixels around here. I would be happy to lend a hand on this, if any help is needed? I would need a 1000ft overview on whats left to make it work, but I would be happy to help.

jpearsonduluth, 2020-11-13:
Any update on this? Been checking out a couple times a month for about a year. Thanks!

adrianstevens, 2020-11-18:
We don't have NeoPixel support yet but there is an APA102 driver that works well if you're interested in alternatives

SteveStrong, 2020-12-09:
I am willing to help also ,  I will watch this issue to see if there are any changes to the status of neo-pixel support

closed 2023-07-11 by adrianstevens