WildernessLabs/Meadow.Foundation #32: There are no unit tests in Meadow.Foundation

aholmes, 2019-11-28:
There are no unit tests in Meadow.Foundation. Having these test will help ensure consistency and quality in individual development and across the open source ecosystem.

adrianstevens, 2020-01-04:
Thanks @aholmes, I love this idea. We're considering some other automation / validation options as we'll need to test against external peripherals.

I'll leave this open for now and we'll update as we make progress.

Cheers!

aholmes, 2020-01-06:
@adrianstevens can you speak more to the other options? Testing against external peripherals sounds like integration testing. The intent with this commit is purely unit tests, which can mock such dependencies. The goal is to test and validate logic and code behavior, rather than physical or external behaviors. Integration and other automation often lives side-by-side with unit tests.

adrianstevens, 2020-12-31:
Here's the test suite the team is building to support Meadow https://github.com/WildernessLabs/Meadow.TestSuite 

I'm going to close this issue but please feel free to continue the conversation there

Cheers!

closed 2020-12-31 by adrianstevens