WildernessLabs/Meadow.Foundation #35: x74595 is just a copy of netduino code

gasia89, 2019-12-04:
It was mentioned on the forum that there was working going on to support this IC, however, its been a few weeks, and I have not heard anything. Is the work active? I would be happy to help move that along if I any help is needed. Is there a branch I can reference for this?

adrianstevens, 2020-01-04:
Thx @gasia89, all of the latest code is in the master branch. We'll likely work on this driver in Jan/Feb. The trick will be setting up the driver to be able to return an IDigitalOutputPort for each pin on the shift register. 

We'll leave this issue open and update as we progress. 

Cheers!

adrianstevens, 2020-01-17:
@gasia89 we've just pushed in updated and working code for the x74595 driver. The nuget package isn't published yet but you can build and test now.

Let us know if you have feedback - and thanks again!

gasia89, 2020-01-24:
Excellent, I will pull master down and give it a go.

closed 2020-01-24 by gasia89