WildernessLabs/Meadow.Foundation #46: Sensors.Temperature.AnalogTemperature_Sample Issue (LM35)

rsforbes, 2020-02-23:
https://github.com/WildernessLabs/Meadow.Foundation/blob/master/Source/Meadow.Foundation.Core.Samples/Sensors.Temperature.AnalogTemperature_Sample/MeadowApp.cs

Using the sample above. My readings are coming back at  

```Temp changed by a degree; new: 21938.23, old: 21712.65
 Temp Changed, temp: 22010.74??C
Temp changed by a degree; new: 22010.74, old: 21938.23
 Temp Changed, temp: 22163.82??C
Temp changed by a degree; new: 22163.82, old: 22010.74
 Temp Changed, temp: 22091.31??C```

In addition, if I remove all pins to the sensor I receive the following:

```Temp changed by a degree; new: 14614.75, old: 14711.42
 Temp Changed, temp: 14912.84??C
Temp changed by a degree; new: 14912.84, old: 14614.75
 Temp Changed, temp: 15114.26??C
Temp changed by a degree; new: 15114.26, old: 14912.84
 Temp Changed, temp: 15589.6??C```

Thoughts?

brucecrombie, 2020-03-22:
It looks like there may be an issue with the VoltageToTemperature method in Meadow.Foundation.Sensors.Temperature.AnalogTemperature.

        protected float VoltageToTemperature(float voltage)
        {
            var yAdjusted = voltage * 1000 - _yIntercept;           
            return yAdjusted * _millivoltsPerDegreeCentigrade * 10;
        }

 y = mx + b  therefore x = (y - b)/m
y = yAdjusted = voltage *1000mV/V
m = _millivoltsPerDegreeCentrigrade
b = _yIntercept

method should be:
       protected float VoltageToTemperature(float voltage)
        {
            var yAdjusted = voltage * 1000;           
            return (yAdjusted  - yIntercept) / _millivoltsPerDegreeCentigrade;
        }

jbfraser1, 2020-04-02:
The proposed fix returns temp in degrees, but other samples expect the return values to be in thousandths of a degree Celsius, such as:
https://www.hackster.io/wilderness-labs/build-your-own-temperature-monitor-with-meadow-edc696

```
protected float VoltageToTemperature(float voltage)
{
   var yAdjusted = voltage * 1000;
   return (yAdjusted - yIntercept) * 1000 / _millivoltsPerDegreeCentigrade;
}
```
Since it is a float, accuracy isn't a problem either way.

For many of the default sensors, where _millivoltsPerDegreeCentigrade is ten, these two methods will return the same value, but for TMP37 or custom calibrations, the proposed method is correct. (Perfect place for some unit tests.)
Current code gives 200000(=200degC) at 1.0V from a TMP37, but proposed code and spec sheet agree that this should be 50degC.

Merge request incoming, keeping the extra three decimal places, but using better calc.

commit 90947d40 (2020-04-04): Fixes #46

calculation now correct for sensors that don't use 10mV per degCelsius.
Still returns degrees Celsius * 1000.
commit e93b382d (2020-04-04): Merge pull request #69 from jbfraser1/bugfix/46VtoTempCalc

Fixes #46 Voltage to temp calculation issue