WildernessLabs/Meadow.Foundation #52: [CharacterDisplay] Displaying shorter lines leaves prior chars on end

patridge, 2020-03-09:
If I display a line of text on the character display and then display a shorter line, any text displayed after the shorter line will not be removed.

## Repro

Accurate of Meadow.Foundation.Displays.Lcd.CharacterDisplay v0.7.6:

```c#
CharacterDisplay display;
public MeadowApp()
{
    Initialize();
    DisplayCharacters();
}

void Initialize()
{
    display = new CharacterDisplay(Device, Device.Pins.D05, Device.Pins.D07,
        Device.Pins.D08, Device.Pins.D09, Device.Pins.D10, Device.Pins.D11,
        16, 2);
}

private void DisplayCharacters()
{
    display.WriteLine("0000000000", 0);
    display.WriteLine("#####", 0);
    // Line 0 will now display `#####00000`
}
```

thegreatco, 2020-04-14:
Hi @patridge,

The `CharacterDispay` included in the hack kit is 20 columns wide, not 16. Inside `WriteLine`, we pad the string with spaces to blank the rest of characters based on the width. If the driver thinks there are only 16 columns, the last 4 won't get blanked.

In `0.7.9` we actually trim the output to the width so long lines don't spill over to other lines, so the proper screen width is more important. We also updated the default constructor parameters to include the correct width.

-Pete

patridge, 2020-04-14:
Well, that sounds like all sorts of things wrong on my part. Thanks for catching all that. I'll poke at things a little more but feel free to close this. I can always re-open or re-submit if I see an issue after correcting my own code.

patridge, 2020-04-14:
Also, that's exciting about the spill-over change. That was definitely something I encountered a couple times. That would make it much easier to diagnose my own issue, too.

adrianstevens, 2020-08-15:
Thanks @patridge!  - closing issue

closed 2020-08-15 by adrianstevens