WildernessLabs/Meadow.Foundation #58: Meadow.Foundation EventHandler does not work

Volirvag49, 2020-03-22:
trying to reproduce this guide. But events are not processed
[http://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Buttons.PushButton.html](url)

Sample code to demonstrate

        using System;
        using System.Threading;
        using Meadow;
        using Meadow.Devices;
        using Meadow.Foundation.Sensors.Buttons;

        namespace MeadowApplication1
        {
            public class MeadowApp : App<F7Micro, MeadowApp>
            {
                protected PushButton pushButton;

                public MeadowApp()
                {
                    Console.WriteLine("Initializing...");

                    pushButton = new PushButton(Device, Device.Pins.D02);

                    //while (true)
                    //{
                    //    Console.WriteLine($"state: {pushButton.State}");
                    //    Thread.Sleep(500);
                    //}

                    pushButton.PressStarted += (s, e) =>
                    {
                        Console.WriteLine("Press started");
                    };
                    pushButton.PressEnded += (s, e) =>
                    {
                        Console.WriteLine("Press ended");
                    };
                    pushButton.Clicked += (s, e) =>
                    {
                        Console.WriteLine("Button Clicked");
                    };
                    pushButton.LongPressClicked += (s, e) =>
                    {
                        Console.WriteLine("Long pressed!");
                    };

                    Console.WriteLine("PushButton ready...");
                }
            }
        }

the commented block with the loop is working fine.
The same thing happens with other examples where EventHandler are used.

magohl, 2020-03-28:
Are you running 3.8? I had the exact same problem using 3.8 and can confirm that since upgrading to 3.9 just released the interrupts and events work.

Volirvag49, 2020-03-28:
> Are you running 3.8? I had the exact same problem using 3.8 and can confirm that since upgrading to 3.9 just released the interrupts and events work.

now it works

bryancostanich, 2020-04-05:
thanks for confirming. this was a regression. :(

closed 2020-04-05 by bryancostanich