WildernessLabs/Meadow.Foundation #70: RotaryEncoder (and RotaryEncoderWithButton) is not triggering on rotation.

airmiles13, 2020-04-03:
With the Rotary Encoder with Button hooked up as per the demo circuit (+ -> 3V3; Gnd-> Gnd; SW->D05, DT-> D06; CLK -> D07) and the below set-up code the value event does not fire properly and the button ended and started events are reversed. The value readings from this are random and do not seem to be dependent on the direction.

` private void InitializeRotaryButton()
        {
            rotaryEncoderWithButton = new RotaryEncoderWithButton(Device, Device.Pins.D07, Device.Pins.D06, Device.Pins.D05, 20);

            rotaryEncoderWithButton.Rotated += (s, e) =>
            {
                if (e.Direction == Meadow.Peripherals.Sensors.Rotary.RotationDirection.Clockwise)
                { value++; }
                else
                { value--; }
                updateLeds();
                Console.WriteLine($"Value = {value}"); 
            };

            rotaryEncoderWithButton.Clicked += (s, e) =>
            {
                value = 0;
                rgbPwmLed.Stop();
            };

            rotaryEncoderWithButton.PressEnded += (s, e) =>
            {
                Console.WriteLine("Press ended");
            };
            rotaryEncoderWithButton.PressStarted += (s, e) =>
            {
                Console.WriteLine($"Value = {value}"); // ; A-Phase Port Reading: {rotaryEncoderWithButton.Results.APhase}; B-Phase Port Reading: {rotaryEncoderWithButton.Results.BPhase}");
                Console.WriteLine("Press started");
            };
        }`

The fix proposed in Pull Request #34  fixes the button issue.  However, when I was addressing the Rotary issue I encountered some strange interactions. First I read the Digital ports my self and tried debugging, everything seems fine with the circuit in logic with events triggering how you would expect when using 2-bit gray code (clockwise aPhase triggers first; counterclockwise bPhase triggers first)

`        private void DebugRotaryButton()
        {
            aPhaseTest = Device.CreateDigitalInputPort(Device.Pins.D07, InterruptMode.EdgeBoth, ResistorMode.PullUp, 200);
            bPhaseTest = Device.CreateDigitalInputPort(Device.Pins.D06, InterruptMode.EdgeBoth, ResistorMode.PullUp, 200);

            aPhaseTest.Changed += APhaseTest_Changed;
            bPhaseTest.Changed += BPhaseTest_Changed;
        }
`
Sample output is attached for this debug code. Rotate clockwise 5 times and counterclockwise 5 times. I will follow-up with some code that found part of the issue... 

[DebugOutput-RotaryEncoderWithButton.txt](https://github.com/WildernessLabs/Meadow.Foundation/files/4427330/DebugOutput-RotaryEncoderWithButton.txt)

airmiles13, 2020-04-03:
To debug this issue I made a new constructor for the RotaryEncoder where I send the debug digital ports so I can see what is happening on the App level more clearly. 

`        public RotaryEncoderWithButton(IIODevice device, IDigitalInputPort aPhasePin, IDigitalInputPort bPhasePin, IPin buttonPin, ResistorMode resistor, int debounceDuration = 20 )
    : base(aPhasePin, bPhasePin)
        {
            _button = new PushButton(device, buttonPin, resistor, debounceDuration);

            _button.Clicked += ButtonClicked;
            _button.PressEnded += ButtonPressEnded;
            _button.PressStarted += ButtonPressStarted;
        }
`
I also split up the PhasePinChanged event into two events in RotaryEncoder. (Note: The logic in the process rotation results actually only works if the B-Phase event is calling it)

`       public RotaryEncoder(IDigitalInputPort aPhasePort, IDigitalInputPort bPhasePort)
        {
            APhasePort = aPhasePort;
            BPhasePort = bPhasePort;

            APhasePort.Changed += PhasePinAChanged; // only need to monitor one of the ports
            BPhasePort.Changed += PhasePinBChanged;
        }

        private void PhasePinAChanged(object sender, DigitalInputPortEventArgs e)
        {
             //This Event is never triggering!
            _results.APhase = APhasePort.State;
            _results.BPhase = BPhasePort.State;

            ProcessRotationResults();
        }

        private void PhasePinBChanged(object sender, DigitalInputPortEventArgs e)
        {

            _results.APhase = APhasePort.State;
            _results.BPhase = BPhasePort.State;

            ProcessRotationResults();
        }

        /// <summary>
        /// Determines the direction of rotation when the PhasePinChanged event is triggered
        /// </summary>
        protected void ProcessRotationResults()
        {
            //Only works if bPhase triggers
            //If BPhase is equal to APhase then the A Phase has changed first and it is rotating clockwise
            OnRaiseRotationEvent(_results.APhase == _results.BPhase ? RotationDirection.Clockwise : RotationDirection.CounterClockwise);
         }
`
When running this code the values of rotation and button clicks are updating properly but the aPhase events never trigger! So, yes the code works for the RotaryButton but there is an issue with the aPhase event never triggering? Does anyone have any ideas why this might be?

[DebugOutput-RotaryEncoderWithButton-2.txt](https://github.com/WildernessLabs/Meadow.Foundation/files/4427463/DebugOutput-RotaryEncoderWithButton-2.txt)

airmiles13, 2020-04-08:
I think the biggest issue is the current event logic doesn't work if there are any incorrect events being thrown. These type of Rotary encoders are very noisy so some extra protection needs to be put in the software. 

I am testing some event-based set-ups to confirm this. It also seems the CLK (aPhase pin) is causing many more false events. One solution may be to only monitor events on the DATA (bPhase pin). I will update a solution with only the bPhase monitored and check it out.

jorgedevs, 2020-10-29:
Hello,

Following up on pending filed issues, I'm going to close this one for now, as both RotaryEncoder and RotaryEncoderWithButton are working in the last released version (0.19.0). Feel free to reopen this or create a new one related to rotary encoders.

closed 2020-10-29 by jorgedevs