WildernessLabs/Meadow.Foundation #81: Using PushButton with Built in RTC causes hang after 17minutes with LCD Display

CartBlanche, 2020-04-18:
Versions I'm using:
Beta 10, 
MeadowFoundation 0.14.1
Meadow.Foundation.Displays.Lcd.CharacterDisplay 0.7.9

Simplest code that demonstrates the issue:
https://gist.github.com/CartBlanche/0186f282679ff637cc4fbea2a9104df1

Steps to reproduce:
Use the code found above. Deploy to the Meadow with the 20x4 LCD attached and confirgured.
After 17minutes the display stops updating.

Comment out the line 
`var downButton = new PushButton(Device, Device.Pins.D12);`

And the  code will continue for hours.

adrianstevens, 2020-08-15:
Thanks @CartBlanche - this should be fixed in b3.11. @jorgedevs do you have a 20x4 display wired up? If not I'll test next week.

jorgedevs, 2020-08-15:
I will test this today. I believe this has been resolved even in 3.12 with the major stability improvements.

jorgedevs, 2020-10-29:
Hello,

I'm following up on this reported issue, and I see the app is much more stable in this latest beta version (0.4.0). I'm closing it in the mean time, but if we have more reported issues on stability, we can re-open it to investigate.

closed 2020-10-29 by jorgedevs