WildernessLabs/Meadow.Foundation #82: Peripheral I2c address check not advisable

SupraBitKid, 2020-04-26:
The following Sensors have address range checks in their constructors:

- SHT31D
- TSL2561
- ADXL345
- BNO055

While the address check is helpful in the simple cases; it is possible to put these sensors on the i2c bus at different addresses.
Linear Device (part of Analog) has a chip to change addresses when there are conflicts on the bus.
see: [LTC4316](https://www.analog.com/en/products/ltc4316.html)

adrianstevens, 2020-05-31:
@SupraBitKid thank you! This has been fixed in the developer branch and the updated drivers will publish with the next Meadow update (likely labelled beta 0.4.0).

https://github.com/WildernessLabs/Meadow.Foundation/commit/1cab8fe1fa23f4397ca901fe78ac9fc9257dc2da

closed 2020-05-31 by adrianstevens

elexisvenator, 2020-05-31:
@SupraBitKid I put together a helper class that can work around any drivers that expect a specific port range, might be useful in the future if more drivers are implemented with similar restrictions

https://gist.github.com/elexisvenator/03c06bee7ce4da0e7138fbbb492085bc