WildernessLabs/Meadow.Foundation #84: Improvment - to DrawBitmap parameter checking

MarkAlanJones, 2020-05-02:
As a user of the library 
I would expect the width and height parameters to be in pixels

as currently implemented - it only works for 1 bit per pixel bit maps, 
so there are 8 pixels per byte horizontally

the user must know to divide the width by 8 and provide an array that is padded to the next byte multiple.

https://github.com/WildernessLabs/Meadow.Foundation/blob/develop/Source/Meadow.Foundation.Libraries_and_Frameworks/Displays.GraphicsLibrary/GraphicsLibrary.cs

            if ((width * height) != bitmap.Length)
            {
                throw new ArgumentException("Width and height do not match the bitmap size.");
            }

There is also drawbitmap at the display level
https://github.com/WildernessLabs/Meadow.Foundation/blob/develop/Source/Meadow.Foundation.Peripherals/Displays.TftSpi/Driver/Displays.TftSpi/DisplayTftSpiBase.cs

and maybe others

MarkAlanJones, 2020-05-02:
Also - consider how would one draw a bitmap less then 8 pixels wide ?

adrianstevens, 2020-05-03:
Thanks! I think we'd just need to pad the buffer, I'll see if I can squeeze that into the next release

adrianstevens, 2020-07-09:
Thanks again @MarkAlanJones - changed the DrawBitmap methods to take the width/height in pixels instead of bytes 

fixed here https://github.com/WildernessLabs/Meadow.Foundation/commit/eb83c5fddc2e563009ca17c834d24a9a8eef8adf

closed 2020-07-09 by adrianstevens