WildernessLabs/Meadow.Foundation #91: DS18B20 Support

mprattinger, 2020-05-23:
Hi,

i'm new to the meadow board and i'm trying to read the temperature of my DS18B20 sensor. I found in the repo an class but this was commented out. Are there any plans to support this type of sensor and the 1-wire protocol?

Thank you in advance,
Michael

adrianstevens, 2020-05-31:
Hi @mprattinger - yes we will be adding support in the future, the limitation today is we don't have a 1-wire bus class/protocol implemented. I think @ctacke has this somewhere on his ToDo list but he's currently prioritizing new Meadow OS features.

Cheers!

mprattinger, 2020-06-18:
Thanks for the Info.
R Michael

closed 2020-06-18 by mprattinger

atrueresistance, 2020-12-15:
I referenced the [docs](https://docs.google.com/spreadsheets/d/1U0asbAYjQgJ6dvqVXnvL8X1BPZiGZcpNl-GsdKagp2s/edit#gid=1827338980) I assume AOT means Ahead Of Time compilation. Is there a timeline or beta release target?