WildernessLabs/Meadow.Foundation #107: OledWing PushButtons internal pullup resistors not set

MarkAlanJones, 2020-08-09:
Problem - new PushButton defaults the resistor to disabled, even if it was set on the pin previously

so the resistor mode must be specified when PushButton is declared

Not sure if declaring an internal resistor when you already have an external one is a good idea or not
button b does not need the resistor, so i left it off.

Fixed code:
```
 public OLED128x32Wing(II2cBus i2cBus, IDigitalInputPort portA, IDigitalInputPort portB, IDigitalInputPort portC)
        {
            this.i2cBus = i2cBus;
            Display = new Ssd1306(this.i2cBus, 0x3C, Ssd1306.DisplayType.OLED128x32);

            ButtonA = new PushButton(portA, ResistorMode.PullUp);
            ButtonB = new PushButton(portB); // has physical resistor
            ButtonC = new PushButton(portC, ResistorMode.PullUp);
        }
```

MarkAlanJones, 2020-08-09:
P.S.

It might be useful for IButton to expose the underlying IPin 
currently you can't see what values are set e.g. ResistorMode

MarkAlanJones, 2020-08-09:
you can get at the IDigitalInputPort - but it is messy

`IDigitalInputPort dic = ((PushButton)oledWing.ButtonC).DigitalIn;`

adrianstevens, 2020-08-15:
Thanks @MarkAlanJones - I made the changes in the WIP branch.

closed 2020-09-27 by MarkAlanJones