WildernessLabs/Meadow.Foundation #109: For HBridgeMotor pwmFrequency can be specified - but 1600 is always used anyway

MarkAlanJones, 2020-09-27:
`public HBridgeMotor(IPwmPort a1Pin, IPwmPort a2Pin, IDigitalOutputPort enablePin, float pwmFrequency = 1600)
        {
            _pwmFrequency = pwmFrequency;

            _motorLeftPwm = a1Pin;
            _motorLeftPwm.Frequency = 1600;
            _motorLeftPwm.Start();

            _motorRighPwm = a2Pin;
            _motorRighPwm.Frequency = 1600;
            _motorRighPwm.Start();

            _enablePort = enablePin;
        }`

Maybe 1600 is the only good value - then it shouldn't be specified ?

jorgedevs, 2020-09-29:
I'll check this out, you're most likely right, thanks for reporting this @MarkAlanJones ! 👍

MarkAlanJones, 2020-10-12:
Upon further testing - i found with my motors that the 1600 made a noticeable audible tone
the tone is higher until around 10000 - when it is mostly non audible - younger hackers may still hear this
I ended up using 12000.  

Higher frequency should give the motor a more consistent voltage ?

bryancostanich, 2020-11-19:
Might consider using a capacitor at the output to smooth the voltage: https://hackaday.com/2011/06/16/beginner-concepts-using-a-low-pass-filter-to-smooth-pwm-output/

adrianstevens, 2021-08-20:
This is fixed in Meadow.Core - will ship with B5.3 - thx @MarkAlanJones!

adrianstevens, 2021-11-29:
B5.3 has shipped - closing issue

closed 2021-11-29 by adrianstevens