WildernessLabs/Meadow.Foundation #110: RA8876 Display

Vission01, 2020-09-28:
Would appreciate a driver for the RA8876 series of displays!

Example Display:
https://www.buydisplay.com/default/serial-spi-arduino-10-1-tft-lcd-display-shield-ra8876-for-due

This may help:
https://github.com/xlatb/ra8876/wiki/ER-TFTM101-1

Thanks!

adrianstevens, 2020-09-30:
Thanks @Vission01  - I actually bought a couple of these but I haven't dug into it yet. I'm just in the process of moving so I won't have access to them for a couple of weeks. I'll dig into this as soon as I have the displays unpacked.

Vission01, 2020-10-26:
@adrianstevens 
OK, please let me know when you have something to try out!

Vission01, 2020-12-08:
@adrianstevens 
Any updates on this driver?