WildernessLabs/Meadow.Foundation #111: Could not restore packages

r-proto, 2020-10-02:
Downloaded the source for Meadow.Foundation and getting could not restore packages error, using Visual Studio for Mac. 

Unable to satisfy conflicting requests for 'Meadow.Foundation': Meadow.Foundation (>= 0.18.0) (via project/Meadow.Foundation.Displays.GraphicsLibrary 0.9.14), Meadow.Foundation (>= 0.18.0) (via project/Meadow.Foundation.Displays.Pcd8544 0.6.12) Framework: (.NETFramework,Version=v4.7.2)
Unable to satisfy conflicting requests for 'Meadow': Meadow (>= 0.16.0) (via package/Meadow.Foundation 0.18.0), Meadow (>= 0.16.0) (via package/Meadow.Foundation 0.18.0) Framework: (.NETFramework,Version=v4.7.2)
Unable to satisfy conflicting requests for 'WildernessLabs.Meadow.Assemblies': WildernessLabs.Meadow.Assemblies (>= 0.3.2) (via package/Meadow 0.16.0), WildernessLabs.Meadow.Assemblies (>= 0.3.2) (via package/Meadow 0.16.0) Framework: (.NETFramework,Version=v4.7.2)
An item with the same key has already been added. Key: (Meadow.Foundation.Displays.GraphicsLibrary, 0.9.14)

jorgedevs, 2020-10-08:
I took a look at this and notice that for some affected peripherals, when you remove and re-add the installed packages seems to fix the reference. My hope was to try to check-in those fixes, but Git didnt detect a meaningful change on those files, since im just removing and re-adding the same PackageReferences on the csproj files.

I'll keep looking into this, see if theres a better solution.

jorgedevs, 2020-10-28:
Hi @r-proto,

Wanted to follow up on this issue. I'm going to close it for now, but do feel free to re-open it if you're still having problems restoring NuGet packages. With the [latest release](http://blog.wildernesslabs.co/meadow-b4-0-1-easier-nugets-reliable-deploy/), you shouldn't no longer keep having this issue.

closed 2020-10-28 by jorgedevs