WildernessLabs/Meadow.Foundation #113: Driver Request: BME680

mjjames, 2020-10-11:
Support the Bosch BME680 which extends the BME280 by including VOC gas sensing

mjjames, 2020-10-11:
I'm going to try and implement this now i have my 280 working correctly

bryancostanich, 2020-11-19:
@mjjames this is actually in flight in the [WIP](https://github.com/WildernessLabs/Meadow.Foundation/commits/WIP) branch. Code his [here](https://github.com/WildernessLabs/Meadow.Foundation/tree/fc22de3eb1a9fe97180cc14ea227a14a3c04be5b/Source/Meadow.Foundation.Peripherals/Sensors.Atmospheric.Bme280).

we were having a discussion internally about architecture. I think we've resolved it, but just need to implement: https://gist.github.com/bryancostanich/6eb11ec24eac60176015b6c30a921fe9

mjjames, 2020-11-19:
I've not had time to implement but I can help test and validate any implementation if that's helpful?

bryancostanich, 2020-11-19:
coolio, was just a head's up that it's in progress.

adrianstevens, 2021-12-09:
I've started working on this driver - I'm hoping to ship in B6.1

adrianstevens, 2022-05-08:
Driver has shipped - closing issue

closed 2022-05-08 by adrianstevens