WildernessLabs/Meadow.Foundation #161: Wifi doesn't work NetworkInterfaceCannotBeStarted

crone66, 2021-03-21:
Hello,

I deployed the project sample Wifi-Basic but for some reason the network interface cannot be started.

The meadow board was successfully update with the following commands;
```
meadow --FlashOS
meadow --MonoUpdateRt
meadow --FlashEsp
```

Visual Studio deployment output:
```
Launching application...
 Meadow successfully started MONO  Initialize hardware...
 Mono enabled, will run app.exe Getting list of access points.
 Error getting access points: CannotStartNetworkInterface
 No access points detected.
 Connecting to WiFi Network FritzBox7490-A
 Unhandled Exception: System.AggregateException: One or more errors occurred. (Cannot connect to network: NetworkInterfaceCannotBeStarted) ---> System.Exception: Cannot connect to network: NetworkInterfaceCannotBeStarted
  at WateringApp.MeadowApp.Initialize2 () <0xc091a8b8 + 0x001d0> in <3dc1dc7fdd1a450880f88198fd634139>:0 
   --- End of inner exception stack trace ---
  at System Threading.Tasks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) <0xc0e3b548 + 0x00022> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at System.Threading.Tasks.Task.Wait (System.Int32 millisecondsTimeout, System.Threading.CancellationToken cancellationToken) <0xc0df0c88 + 0x000a0> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at System.Threading.Tasks.Task.Wait ()  0xc0df0a60 + 0x00018> in <fe755e761f0348d69fca7362f35b16b3>:0 
  at WateringApp.MeadowApp..ctor () <0xc083bfd0 + 0x000c8> in <3dc1dc7fdd1a450880f88198fd634139>:0 
  at WateringApp.Program.Main (System.String[] args) <0xc083b8c8 + 0x00034> in <3dc1dc7fdd1a450880f88198fd634139>:0 
---> (Inner Exception #0) System.Exception: Cannot connect to network: NetworkInterfaceCannotBeStarted
  at WateringApp.MeadowApp.Initialize2 () <0xc091a8b8 + 0x001d0> in <3dc1dc7fdd1a450880f88198fd634139>:0 <---

 [ERROR] FATAL UNHANDLED EXCEPTION: System.AggregateException: On e or more errors occurred. (Cannot connect to network: NetworkIn terfaceCannotBeStarted) ---> System.Exception: Cannot connect to  network: NetworkInterfaceCannotBeStarted
   at WateringApp.MeadowApp.Initialize2 () <0xc091a8b8 + 0x001d0>  in <3dc1dc7fdd1a450880f88198fd634139>:0 
    --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional (System.Bool ean includeTaskCanceledExceptions) <0xc0e3b548 + 0x00022> in <fe 755e761f0348d69fca7362f35b16b3>:0 
   at System.Threading.Tasks.Task.Wait (System.Int32 milliseconds Timeout, System.Threading.CancellationToken cancellationToken) < 0xc0df0c88 + 0x000a0> in <fe755e761f0348d69fca7362f35b16b3>:0 
   at System.Threading.Tasks.Task.Wait () <0xc0df0a60 + 0x00018>  in <fe755e761f0348d69fca7362f35b16b3>:0 
   at WateringApp.MeadowApp..ctor () <0xc083bfd0 + 0x000c8> in <3 dc1dc7fdd1a450880f88198fd634139>:0 
   at WateringApp.Program.Main (System.String[] args) <0xc083b8c8  + 0x00034> in <3dc1dc7fdd1a450880f88198fd634139>:0 
 ---> (Inner Exception #0) System.Exception: Cannot connect to ne twork: NetworkInterfaceCannotBeStarted
   at WateringApp.MeadowApp.Initialize2 () <0xc091a8b8 + 0x001d0>  in <3dc1dc7fdd1a450880f88198fd634139>:0 <---
```

NevynUK, 2021-03-21:
Which version of the OS do you have installed?  We have recently released 0.4.6.1 which you should be able to get by issuing the command `meadow --Download`

Did you issue the command `meadow --MonoDisable` ?  The ESP binaries cannot be flashed unless Mono is disabled.

crone66, 2021-03-21:
I Installed everything today and I deactivated Mono like mentioned in the documentation before I executed the following commands:

```
meadow --MonoUpdateRt
meadow --FlashEsp
```

The commandline output looked like everything was installed successfully.

NevynUK, 2021-03-26:
This issue has been fixed in 0.4.6.2.  This was released 25 March 2021.

closed 2021-03-26 by NevynUK