WildernessLabs/Meadow.Foundation #169: Yx5300: `data2` is send in place of `data1`

craigktreasure, 2021-04-26:
Many commands are not working for the Yx5300 driver as a result of an error in the code causing `data2` to be sent in the position of `data1`. See [here](https://github.com/WildernessLabs/Meadow.Foundation/blob/main/Source/Meadow.Foundation.Peripherals/Audio.Mp3.Yx5300/Driver/Audio.Mp3.Yx5300/Yx5300.cs#L214].

```csharp
            // Command Structure 0x7E 0xFF 0x06 CMD FBACK DAT1 DAT2 0xEF
            sendBuffer[0] = 0x7E;    // Start byte
            sendBuffer[1] = 0xFF;    // Version
            sendBuffer[2] = 0x06;    // Command length not including Start and End byte.
            sendBuffer[3] = (byte)command; // Command
            sendBuffer[4] = 0x01;    // Feedback 0x00 NO, 0x01 YES
            sendBuffer[5] = data2;    // DATA1 datah
            sendBuffer[6] = data2;    // DATA2 datal
            sendBuffer[7] = 0xEF;    // End byte
```

`data2` is sent twice. `data1` should be sent for the high (`datah`) portion.

adrianstevens, 2021-05-24:
Thanks @craigktreasure  - I'll get this fixed before the next release

adrianstevens, 2021-05-24:
Fixed for B5.0

closed 2021-07-10 by adrianstevens