WildernessLabs/Meadow.Foundation #180: Ds3231x b5.1 breaking changes

brucecrombie, 2021-07-17:
It looks like as of b5.1 Meadow.Foundation.RTCs.Ds323x is broken.

Meadow.Hardware.IByteCommunications has been changed and ReadRegisters is now obsolete so method DisplayRegisters no longer works as is broken at    var data = ds323x.ReadRegisters(0, 0x12);
Ds323x also uses ReadRegisters in other places as well. I see that WriteRegisnters is also used and has be depreciated.

I gather this is all part of the changes under way to move to Span. Do you have someone on working on cleaning these up now or are you looking for someone to help with work on some of these peripheral drivers and create pull requests.
If you are looking for help I just want to confirm the pattern you are using for modifying this code.
I was thinking of just changing
```
var data = ds323x.ReadRegisters(0, 0x12);
DebugInformation.DisplayRegisters(0, data);
```
to
```
byte[] data = new byte[0x12];
ds323x.ReadRegister(0, new Span<byte>(data));
DebugInformation.DisplayRegisters(0, data);
```
Probably good to get rid of 0x12 and use `private const byte REGISTERS_LENGTH = 0x12;`

adrianstevens, 2021-07-19:
Thanks @brucecrombie I'll look into this today (and sorry about that)

brucecrombie, 2021-07-26:
When you get a chance to look at it, also have a look at line 194
it is:
`switch (InterruptPort.InterruptMode) `and should be `switch (interruptPort.InterruptMode)`

When initializing a new Ds323x object that section of the code is checking the Initialization function parameter interruptPort for the correct mode (rising edge) when a IDigitalInputPort is passed to the Ds323x constructor.
InterruptPort is the IDigitalInputPort property that gets assigned the reference passed into the class through interruptPort once error checking is complete.
Looks like just a typo to me :)

adrianstevens, 2021-08-20:
Thx again @brucecrombie - I'll fix that typo on line 194 now. And sorry for the delay, I haven't been able to find the hardware :) @jorgedevs is going to take a look.

closed 2022-08-04 by adrianstevens