WildernessLabs/Meadow.Foundation #186: ICs.IOExpanders.Is31fl3731_Sample question

Tridy, 2021-08-09:
Hi.

Is there a way to document which pins need to be connected to F7 Micro in order for it to work and if there are any extra components that are required. How did you guys test it when this sample was created? More details on how I connect it is in [this](https://community.wildernesslabs.co/t/how-to-connect-and-work-with-led-shim-is31fl3731/1583) post.

Thanks!

adrianstevens, 2021-08-20:
Hey @Tridy I used it in the Adafruit CharliePlex 15x7 FeatherWing https://www.adafruit.com/product/3163 

It should just require 3.3V, ground, I2C data (pin 7) and I2c clock (pin 8).

If it's not working you may need pull up or pull down resistors on pins 7 & 8 (either work for I2C).

Tridy, 2021-08-20:
I am not sure about 3.3V

the description of LED Shim says:

> Needs 5v power

https://pinout.xyz/pinout/led_shim

adrianstevens, 2021-11-25:
Ah interesting ..... Meadow is 5V tolerant so it should be fine if you power it off of 5V

Tridy, 2021-11-28:
Well, I got tired of trying and gave up. Maybe one day someone could create videos of actual devices wired and working, and not just look like it should work on paper. My curiosity is worn out at this point.

adrianstevens, 2022-05-08:
Just an update - I've used this driver with another peripheral - the Mikro 16x9 Click board and it's working correctly.

That board only needs I2C clock, data, 3.3V and ground.

adrianstevens, 2022-07-12:
closing issue

closed 2022-07-12 by adrianstevens