WildernessLabs/Meadow.Foundation #208: PushButton events on Linux only give Cliecked events

ctacke, 2021-11-07:
Running M4L on Raspberry Pi.

```
void TestExternalResistorButtons()
{
    _pushButtons = new List<PushButton>();

    var inputExternalPullUp = Device.CreateDigitalInputPort(
        pin: Device.Pins.GPIO21,
        InterruptMode.EdgeBoth,
        resistorMode: ResistorMode.ExternalPullUp);
    var buttonExternalPullUp = new PushButton(inputExternalPullUp);

    _pushButtons.Add(buttonExternalPullUp);

    var inputExternalPullDown = Device.CreateDigitalInputPort(
        pin: Device.Pins.GPIO20,
        InterruptMode.EdgeBoth,
        resistorMode: ResistorMode.ExternalPullDown);
    var buttonExternalPullDown = new PushButton(inputExternalPullDown);

    _pushButtons.Add(buttonExternalPullDown);

    foreach (var pushButton in _pushButtons)
    {
        pushButton.LongClickedThreshold = new TimeSpan(0, 0, 1);

        pushButton.Clicked += PushButtonClicked;
        pushButton.PressStarted += PushButtonPressStarted;
        pushButton.PressEnded += PushButtonPressEnded;
        pushButton.LongClicked += PushButtonLongClicked;
    }
}
```
Both external pull-up or external pull-down variations only give Clicked events.  Never see the Press start/end/long click.

adrianstevens, 2022-11-17:
@ctacke low pri but I'd curious to know if this is fixed with the latest changes to PushButton

adrianstevens, 2023-01-25:
Confident this is fixed - closing issue

closed 2023-01-25 by adrianstevens