WildernessLabs/Meadow.Foundation #209: Out of Bounds exception if graphicslibrary is used to draw off screen

MarkAlanJones, 2021-11-10:
I noticed this with several projects when upgrading to beta 6.0 - usually I set the rotation of the graphics library to 270 
and set the display to IgnoreOutofBoundsPixels = true, but setting it to either true or false causes the same results.

e.g. initialization code

            display = new St7789(
                device: Device,
                spiBus: spiBus,
                chipSelectPin: null,
                dcPin: Device.Pins.D01,
                resetPin: Device.Pins.D00,
                width: displayWidth, height: displayHeight);

            display.IgnoreOutOfBoundsPixels = true;

            graphics = new GraphicsLibrary(display)
            {
                Rotation = RotationType._270Degrees,
                CurrentFont = new Font12x16()
            };

Out of bounds caused by

1) graphics.DrawRectangle(0, 0, display.Width, display.Height, Color.White, true)
2) graphics.Clear(Color.White)
3) graphics.DrawCircle(rand.Next(displayWidth), rand.Next(displayHeight), rand.Next(displayHeight / 2) + 1, RandColor(), true)
4) graphics.DrawHorizontalLine(rand.Next(displayWidth), rand.Next(displayHeight), rand.Next(displayWidth), RandColor())
5) graphics.DrawVerticalLine(rand.Next(displayWidth), rand.Next(displayHeight), rand.Next(displayWidth), RandColor())
6) graphics.Drawtext as well - if it goes off the edge.

stack trace looks as follows

Notice the display calls Graphics.Buffers.BufferRgb444.SetPixel which seems to throw the exception.
It seems to report the trace multiple times.

2>[10-Nov-2021 1:50:23 PM] Meadow StdOut: Unhandled Exception: System.IndexOutOfRangeException: Index was outside the bounds of the array.
2>  at Meadow.Foundation.Graphics.Buffers.BufferRgb444.SetPixel (System.Int32 x, System.Int32 y, System.UInt16 color) <0xc0ebe480 + 0x000a8> in <b87bfb2a84704b13845b87e35b9e4fef>:0 
2>  at Meadow.Foundation.Displays.TftSpi.TftSpiBase.SetPixel (System.Int32 x, System.Int32 y, System.UInt16 
2>[10-Nov-2021 1:50:23 PM] Meadow StdOut: olor) <0xc0ebe400 + 0x00048> in <7f21c5ccb2ad40e8b12bbd6b6870aac9>:0 
2>  at Meadow.Foundation.Displays.TftSpi.TftSpiBase.DrawPixel (System.Int32 x, System.Int32 y, Meadow.Foundation.Color color) <0xc0ebe120 + 0x00022> in <7f21c5ccb2ad40e8b12bbd6b6870aac9>:0 
2>  at Meadow.Foundation.Graphics.GraphicsLibrary.DrawPixel (System.Int32 x, System.Int32 y) <0xc0ebddc0 + 0x00034> in <b87bfb2
2>[10-Nov-2021 1:50:23 PM] Meadow StdOut: 84704b13845b87e35b9e4fef>:0 
2>  at Meadow.Foundation.Graphics.GraphicsLibrary.DrawCircleOutline (System.Int32 centerX, System.Int32 centerY, System.Int32 radius, System.Boolean centerBetweenPixels) <0xc0ebda88 + 0x00082> in <b87bfb2a84704b13845b87e35b9e4fef>:0 
2>  at Meadow.Foundation.Graphics.GraphicsLibrary.DrawCircle (System.Int32 centerX, System.Int32 centerY, System.Int32 radiu
2>[10-Nov-2021 1:50:23 PM] Meadow StdErr: [ERROR] FATAL UNHANDLED EXCEPTION: System.IndexOutOfRangeExcepti
2>[10-Nov-2021 1:50:23 PM] Meadow StdOut: , Meadow.Foundation.Color color, System.Boolean filled, System.Boolean centerBetweenPixels) <0xc0ebd948 + 0x0007a> in <b87bfb2a84704b13845b87e35b9e4fef>:0 
2>  at MeadowGraphicsBench.MeadowApp.BenchCircles (System.Int32 num) <0xc0ebaaf8 + 0x000aa> in <e6ed42551f2144e18909cf80c65f97ca>:0 
2>  at MeadowGraphicsBench.MeadowApp..ctor () <0xc08252f8 + 0x0003e> in <e6ed42551f2144e18909cf80c
2>[10-Nov-2021 1:50:23 PM] Meadow StdErr: on: Index was outside the bounds of the array.
2>  at Meadow.Foundation.Graphics.Buffers.BufferRgb444.SetPixel (System.Int32 x, System.Int32 y, System.UInt16 color) <0xc0ebe480 + 0x000a8> in <b87bfb2a84704b13845b87e35b9e4fef>:0 
2>  at Meadow.Foundation.Displays.TftSpi.TftSpiBase.SetPixel (System.Int32 x, System.Int32 y, System.UInt16 color) <0xc0ebe400 + 0x00048> in <7f21c5ccb2ad40e8
2>[10-Nov-2021 1:50:23 PM] Meadow StdOut: 5f97ca>:0 
2>  at MeadowGraphicsBench.Program.Main (System.String[] args) <0xc0824988 + 0x00034> in <e6ed42551f2144e18909cf80c65f97ca>:0 
2>
2>[10-Nov-2021 1:50:23 PM] Meadow StdErr: 12bbd6b6870aac9>:0 
2>  at Meadow.Foundation.Displays.TftSpi.TftSpiBase.DrawPixel (System.Int32 x, System.Int32 y, Meadow.Foundation.Color color) <0xc0ebe120 + 0x00022> in <7f21c5ccb2ad40e8b12bbd6b6870aac9>:0 
2>  at Meadow.Foundation.Graphics.GraphicsLibrary.DrawPixel (System.Int32 x, System.Int32 y) <0xc0ebddc0 + 0x00034> in <b87bfb2a84704b13845b87e35b9e4fef>:0 
2>  at Meadow.Foundati
2>[10-Nov-2021 1:50:23 PM] Meadow StdErr: n.Graphics.GraphicsLibrary.DrawCircleOutline (System.Int32 centerX, System.Int32 centerY, System.Int32 radius, System.Boolean centerBetweenPixels) <0xc0ebda88 + 0x00082> in <b87bfb2a84704b13845b87e35b9e4fef>:0 
2>  at Meadow.Foundation.Graphics.GraphicsLibrary.DrawCircle (System.Int32 centerX, System.Int32 centerY, System.Int32 radius, Meadow.Foundation.Color color, System.Boolean f
2>[10-Nov-2021 1:50:23 PM] Meadow StdErr: lled, System.Boolean centerBetweenPixels) <0xc0ebd948 + 0x0007a> in <b87bfb2a84704b13845b87e35b9e4fef>:0 
2>  at MeadowGraphicsBench.MeadowApp.BenchCircles (System.Int32 num) <0xc0ebaaf8 + 0x000aa> in <e6ed42551f2144e18909cf80c65f97ca>:0 
2>  at MeadowGraphicsBench.MeadowApp..ctor () <0xc08252f8 + 0x0003e> in <e6ed42551f2144e18909cf80c65f97ca>:0 
2>  at MeadowGraphicsBench.Program.Main 
2>[10-Nov-2021 1:50:23 PM] Meadow StdErr: System.String[] args) <0xc0824988 + 0x00034> in <e6ed42551f2144e18909cf80c65f97ca>:0 
2>
2>[10-Nov-2021 1:50:28 PM] Connecting to Meadow on COM3

adrianstevens, 2021-11-13:
Thanks @MarkAlanJones! I'm testing now but I believe I have it fixed.

All of the display drivers now use standard buffer classes for the offscreen buffers which was great because the optimizations from various displays are now common and shared across drivers.

Apparently a few mistakes slipped by in the refactoring ;)

Looks like there was both a bug in the Clear call - the rotation calculations were off by 1 and the IgnoreOutOfBoundsPixels check in the Tft Spi base class was omitted. Both are fixed - I'll see if we can get an out-of-band release on these packages this weekend.

adrianstevens, 2021-11-18:
Quick update - fix has shipped

closed 2021-12-21 by adrianstevens