WildernessLabs/Meadow.Foundation #260: `IDisplayBuffer.displayColorMode` usage?

pekkah, 2022-01-16:
`IDisplayBuffer` has an property called [`displayColorMode`](https://github.com/WildernessLabs/Meadow.Foundation/blob/3f6f9b06e6189892c86059c3a046209b853e2400/Source/Meadow.Foundation.Core/Displays/IDisplayBuffer.cs#L27) this makes is hard to implement your own buffers are you would need to patch `ColorType` enum to contain your buffer format. Is there actual use for this enum and could it be changed to an struct/class which would describe the PixelFormat of the buffer?

adrianstevens, 2022-01-17:
That's an interesting idea - we're using it to convert data between buffer types. So the framework can copy data between buffers of different color depths. 

We could probably maintain the same behavior with a struct.

I'm also curious to know what buffer format(s) you're adding?

pekkah, 2022-01-17:
Hmm I don't see it being used here https://github.com/WildernessLabs/Meadow.Foundation/blob/3f6f9b06e6189892c86059c3a046209b853e2400/Source/Meadow.Foundation.Libraries_and_Frameworks/Graphics/MicroGraphics/Driver/Graphics.MicroGraphics/Buffers/BufferBase.cs#L69

adrianstevens, 2022-02-05:
That's good call - we're actually comparing the buffer type instead of the color depth 

https://github.com/WildernessLabs/Meadow.Foundation/blob/3f6f9b06e6189892c86059c3a046209b853e2400/Source/Meadow.Foundation.Libraries_and_Frameworks/Graphics/MicroGraphics/Driver/Graphics.MicroGraphics/Buffers/BufferBase.cs#L58

adrianstevens, 2022-07-12:
I wonder if we should look into this before v1 .... so what we would need to describe the buffer?

bits per pixel? the bit packing order? do want to support anything other than just RGB? (e.g. the APA102 has an extra byte for brightness, we could support RGBY or RGBW ... _

adrianstevens, 2023-07-11:
Since this is a closed set and doesn't change often we'll leave it as-is for now. Closing issue but feel free to re-open to discuss further.

closed 2023-07-11 by adrianstevens