WildernessLabs/Meadow.Foundation #269: DrawBuffer does not observe rotated graphics

HuskyRiver, 2022-02-24:
When using an Ili9341 display with the rotation to 270, DrawBuffer doesn't rotate the placed bitmap. It is displayed sideways. The DrawBuffer function uses the correct XY coordinates of the rotated graphics object. The upper left corner of the bitmap is located at the specified XY, but the bitmap is rotated 90 degrees clockwise around that XY point. Drawing text rotates as expected

Attached simple code shows the problem.

[DFRotateProb.zip](https://github.com/WildernessLabs/Meadow.Foundation/files/8137050/DFRotateProb.zip)
.

adrianstevens, 2022-02-25:
Thanks @HuskyRiver  - my apologies, we have fixed this in the developer branch and we'll ship new nuget packages with the next release 

https://github.com/WildernessLabs/Meadow.Foundation/issues/242

HuskyRiver, 2022-02-28:
Great, thanks!

closed 2022-03-07 by adrianstevens