WildernessLabs/Meadow.Foundation #297: Filled Rounded Rectangle Out of Bounds exception

MarkAlanJones, 2022-03-19:
in general if you set  display.IgnoreOutOfBoundsPixels = true; 
then you will not get an out of bounds exception if you draw a shape that goes outside the screen boundaries

this is not the case for Rounded Rectangles that are filled  
(works ok for non rounded rectangles, circles and triangles, both filled and unfilled - only filled rounded rectangles seem to be an issue)

e.g. 
 ```
                int x = rand.Next(displayWidth);
                int y = rand.Next(displayHeight);
                int w = rand.Next(displayWidth / 4);
                int h = rand.Next(displayHeight / 4);
                Console.WriteLine($"{i} Rounded Filled Rectangle {w}x{h} @ {x}.{y}");
                // crashes here argumentoutofrange 
                graphics.DrawRoundedRectangle(x, y, w, h, 5, RandColor(), true);
``
[MeadowTestRoundedRect.zip](https://github.com/WildernessLabs/Meadow.Foundation/files/8309778/MeadowTestRoundedRect.zip)
`

MarkAlanJones, 2022-03-19:
stack trace

`[19-Mar-2022 1:27:59 PM] Meadow StdOut: Unhandled Exception: System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values.
  at Meadow.Foundation.Graphics.Buffers.BufferRgb444.Fill (Meadow.Foundation.Color color, System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height) <0xc0ee4548 + 0x00046> in <a45bb49756af40ce87fb8c3c556680ba>:0 
  at Meadow.Foundation.Displays.TftSpi.Tft[19-Mar-2022 1:27:59 PM] Meadow StdOut: piBase.Fill (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, Meadow.Foundation.Color color) <0xc0ee43e0 + 0x0008c> in <65d2927e712a43a2afd3ba5e66808aad>:0 
  at Meadow.Foundation.Graphics.MicroGraphics.DrawRectangle (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, Meadow.Foundation.Color color, System.Boolean filled) <0xc0ee4028 + [19-Mar-2022 1:27:59 PM] Meadow StdErr: [ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentOutOfRangeExce[19-Mar-2022 1:27:59 PM] Meadow StdOut: x00170> in <a45bb49756af40ce87fb8c3c556680ba>:0 
  at Meadow.Foundation.Graphics.MicroGraphics.DrawRoundedRectangle (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, System.Int32 cornerRadius, Meadow.Foundation.Color color, System.Boolean filled) <0xc0edffc0 + 0x001b8> in <a45bb49756af40ce87fb8c3c556680ba>:0 
  at MeadowTestRoundedRect.MeadowApp.BenchRoundR[19-Mar-2022 1:27:59 PM] Meadow StdErr: ption: Specified argument was out of the range of valid values.
[19-Mar-2022 1:27:59 PM] Meadow StdOut: ct (System.Int32 num) <0xc0ed0118 + 0x0029e> in <402509c55c2348c4b5cb4f034c19d3c4>:0 
  at MeadowTestRoundedRect.MeadowApp..ctor () <0xc0835f98 + 0x00044> in <402509c55c2348c4b5cb4f034c19d3c4>:0 
  at MeadowTestRoundedRect.Program.Main (System.String[] args) <0xc08359e0 + 0x00034> in <402509c55c2348c4b5cb4f034c19d3c4>:0 
[19-Mar-2022 1:27:59 PM] Meadow StdErr:   at Meadow.Foundation.Graphics.Buffers.BufferRgb444.Fill (Meadow.Foundation.Color color, System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height) <0xc0ee4548 + 0x00046> in <a45bb49756af40ce87fb8c3c556680ba>:0 
  at Meadow.Foundation.Displays.TftSpi.TftSpiBase.Fill (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, Meadow.Foundation.Color col[19-Mar-2022 1:27:59 PM] Meadow StdErr: r) <0xc0ee43e0 + 0x0008c> in <65d2927e712a43a2afd3ba5e66808aad>:0 
  at Meadow.Foundation.Graphics.MicroGraphics.DrawRectangle (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, Meadow.Foundation.Color color, System.Boolean filled) <0xc0ee4028 + 0x00170> in <a45bb49756af40ce87fb8c3c556680ba>:0 
  at Meadow.Foundation.Graphics.MicroGraphics.DrawRoundedRectang[19-Mar-2022 1:27:59 PM] Meadow StdErr: e (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, System.Int32 cornerRadius, Meadow.Foundation.Color color, System.Boolean filled) <0xc0edffc0 + 0x001b8> in <a45bb49756af40ce87fb8c3c556680ba>:0 
  at MeadowTestRoundedRect.MeadowApp.BenchRoundRect (System.Int32 num) <0xc0ed0118 + 0x0029e> in <402509c55c2348c4b5cb4f034c19d3c4>:0 
  at MeadowTestRoundedRect.[19-Mar-2022 1:27:59 PM] Meadow StdErr: eadowApp..ctor () <0xc0835f98 + 0x00044> in <402509c55c2348c4b5cb4f034c19d3c4>:0 
  at MeadowTestRoundedRect.Program.Main (System.String[] args) <0xc08359e0 + 0x00034> in <402509c55c2348c4b5cb4f034c19d3c4>:0 
`

adrianstevens, 2022-03-20:
I suspect you're generating width and height values that smaller than twice the corner radius - so internally it's attempting to draw a rectangle with a negative width/height

adrianstevens, 2022-03-20:
That should do it - thx as always @MarkAlanJones  :)

Fixed for B6.4

MarkAlanJones, 2022-06-25:
Using Beta 6.4 I find the Rounded Rectangles are ok, but the regular rectangles are now throwing an exception

`1>[25-Jun-2022 3:15:05 PM] Meadow StdOut: Unhandled Exception: System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values.
1>  at Meadow.Foundation.Graphics.Buffers.BufferRgb444.Fill (Meadow.Foundation.Color color, System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height) <0xc0ef0788 + 0x00046> in <87fa33887e5b4d81b8b5236704765aed>:0 
1>  at Meadow.Foundation.Displays.TftSpi.Tft
1>[25-Jun-2022 3:15:05 PM] Meadow StdOut: piBase.Fill (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, Meadow.Foundation.Color color) <0xc0ef0620 + 0x0008c> in <edcb9c941c914cd69e265efa2ad06c0a>:0 
1>  at Meadow.Foundation.Graphics.MicroGraphics.DrawRectangle (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, Meadow.Foundation.Color color, System.Boolean filled) <0xc0eedc38 + 
1>[25-Jun-2022 3:15:05 PM] Meadow StdErr: [ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentOutOfRangeExce
1>[25-Jun-2022 3:15:05 PM] Meadow StdOut: x001b8> in <87fa33887e5b4d81b8b5236704765aed>:0 
1>  at MeadowGraphicsBench.MeadowApp.BenchRect (System.Int32 num) <0xc0ee70a8 + 0x001ea> in <d576f89883ef417290be169b7293299a>:0 
1>  at MeadowGraphicsBench.MeadowApp..ctor () <0xc0838468 + 0x00044> in <d576f89883ef417290be169b7293299a>:0 
1>  at MeadowGraphicsBench.Program.Main (System.String[] args) <0xc0837e40 + 0x00034> in <d576f89883
1>[25-Jun-2022 3:15:05 PM] Meadow StdErr: ption: Specified argument was out of the range of valid values.
1>  at Meadow.Foundation.Graphics.Buffers.BufferRgb444.Fill (Meadow.Foundation.Color color, System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height) <0xc0ef0788 + 0x00046> in <87fa33887e5b4d81b8b5236704765aed>:0 
1>  at Meadow.Foundation.Displays.TftSpi.TftSpiBase.Fill (System.Int32 x, System.Int32 y, Syst
1>[25-Jun-2022 3:15:05 PM] Meadow StdOut: f417290be169b7293299a>:0 
1>
1>[25-Jun-2022 3:15:05 PM] Meadow StdErr: m.Int32 width, System.Int32 height, Meadow.Foundation.Color color) <0xc0ef0620 + 0x0008c> in <edcb9c941c914cd69e265efa2ad06c0a>:0 
1>  at Meadow.Foundation.Graphics.MicroGraphics.DrawRectangle (System.Int32 x, System.Int32 y, System.Int32 width, System.Int32 height, Meadow.Foundation.Color color, System.Boolean filled) <0xc0eedc38 + 0x001b8> in <87fa33887e5b4d81b8b5236704765aed>:0 
1>
1>[25-Jun-2022 3:15:05 PM] Meadow StdErr:  at MeadowGraphicsBench.MeadowApp.BenchRect (System.Int32 num) <0xc0ee70a8 + 0x001ea> in <d576f89883ef417290be169b7293299a>:0 
1>  at MeadowGraphicsBench.MeadowApp..ctor () <0xc0838468 + 0x00044> in <d576f89883ef417290be169b7293299a>:0 
1>  at MeadowGraphicsBench.Program.Main (System.String[] args) <0xc0837e40 + 0x00034> in <d576f89883ef417290be169b7293299a>:0 
1>
`

it is FILLED rectangle

```
           graphics.Clear();
            for (int i = 1; i < num; i++)
            {
                graphics.DrawRectangle(rand.Next(displayWidth), rand.Next(displayHeight), rand.Next(displayWidth / 4), rand.Next(displayHeight / 4), RandColor(), filled: true);
            }
            graphics.Show();
```

MarkAlanJones, 2022-06-25:
MicroGraphics Library 0.9.54

commit c9a3603b (2022-03-20): MicroGraphics XML comments + guardrails on width/height for DrawRectangle - fixes #297
closed 2022-07-11 by adrianstevens