WildernessLabs/Meadow.Foundation #316: DatatLogger ThinkSpeak package ID is incorrect

brucecrombie, 2022-04-23:
Package ID for Meadow.Foundation/Source/Meadow.Foundation.Libraries_and_Frameworks/DataLoggers/DataLoggers.ThingSpeak is incorrect. I currently is set to  `<PackageId>Meadow.Foundation.DataLoggers.AdafruitIO</PackageId>`
It should be set to   `<PackageId>Meadow.Foundation.DataLoggers.ThinkSpeak</PackageId>` to correctly match the class name.

Currently this has the affect of overwriting the AdafruitIO package. When you install the nuget package for AdafruitIO you instead get the ThinkSpeak class installed.

adrianstevens, 2022-04-23:
Thanks Bruce! - I just opened PR to fix - we'll have the corrected with B6.4

https://github.com/WildernessLabs/Meadow.Foundation/pull/317

brucecrombie, 2022-04-24:
Thanks Adrian.

PR #317 (2022-04-23): Fix ThingSpeak package ID
fixes #316