WildernessLabs/Meadow.Foundation #333: MicroGraphics should support renderers that extend it's capabilities

profexorgeek, 2022-05-10:
**TLDR** MicroGraphics should:

1. make its `private readonly IGraphicsDisplay display` field `protected` instead of `private`
2. Add the `virtual` keyword to the `Clear` method so it can be overridden*
3. Add the `virtual` keyword to the `Show` method so it can be overridden*

> *NOTE: I don't know what impact virtual lookups have for your implementation of .NET but you are already using virtual methods in "frame time" in other places.

**Detailed Why**

I am building a tiny 2D game engine, which is hosted here for reference:
https://github.com/profexorgeek/Glade2d

I want my renderer to have  these capabilities:

1. Render to a lower-resolution buffer and then draw the software buffer with scaling to the driver display
1. Render a frame from a larger spritesheet
1. Render with transparency
1. Overlay an FPS counter (or other debug information)
1. Texture buffer management

Most of these can be added fairly easily. However, implementing any kind of double-buffering or half-resolution rendering is impossible because MicroGraphics is hardcoded to draw to the driver's graphics buffer. Untangling this is difficult because the display object is both private and readonly. To work around this I had to:

- Create a ["fake" `IGraphicsDisplay` implementation that wraps the actual `IGraphicsDisplay` driver instance](https://github.com/profexorgeek/Glade2d/blob/master/Glade2d/Graphics/GraphicsDisplayBufferRgb888.cs) and cascades calls with scaling
- Create a renderer with a [very hacky factory method and private constructor](https://github.com/profexorgeek/Glade2d/blob/master/Glade2d/Graphics/Renderer.cs#L39) in order to have access to both buffers.

I was able to successfully create a renderer which can render a complex scene graph. Given this spritesheet:
[image]

The renderer can composite a layered frame composed of many sprites and render it at half resolution (2x scale)
[image]

If you want to test this yourself, the repo link is at the top of this issue.

profexorgeek, 2022-05-10:
The more I think about this, the more important I think it is to decouple all of the Draw methods from the display device. These methods should probably act on a `IDisplayBuffer targetBuffer` object that is the display device buffer by default, but can be changed. Instead of calling things like `device.DrawPixel` they should call `targetBuffer.DrawPixel` which allows the dev to have more control over where the renderer is drawing.

This would allow drawing to temporary buffers to cache complex draws. For example, this would probably make the clock demo faster because you could composite the watch face and hands as separate buffers and just composite them from the stored buffers every tick.

adrianstevens, 2022-05-10:
@profexorgeek thank you!

PR is here: https://github.com/WildernessLabs/Meadow.Foundation/pull/334/files

And I love the idea of updating MicroGraphics to operate on a buffer instead of the display itself. 

Do you want to sync and talk it through?

adrianstevens, 2022-05-10:
The only issue we have today is - not every display holds an IDisplayBuffer object (but most do ... I think there's only 2 or 3). But I believe we could either update the driver or the driver could implement IDisplayBuffer directly ....

profexorgeek, 2022-05-10:
Hey, I didn't expect this level of responsiveness, this is awesome and the PR looks great. A couple things:

- I'd be happy to sync up, I'm an EE newbie and there are a lot of perf considerations here that I'm ignorant of.
- I work in Mountain Time and could slack huddle most days and some evenings. I have some time this afternoon, otherwise I'm heading out tomorrow for a conference where I hope to do some microcontroller hacking.
- The API for a buffer and a display are very similar IIRC - I wonder if this abstraction could be combined so a renderer could behave ignorant of its underlying device? I suspect you'll have a lot more clarity on the finer points here
- My Glade2d project adds some unique features in terms of API but is... not performant. I'd be happy to move those features into core MeadowGraphics if you think they'd be broadly useful

Feel free to DM me on Slack to discuss more, this is neat stuff!

closed 2022-05-26 by adrianstevens