WildernessLabs/Meadow.Foundation #338: Meadow drawing API

profexorgeek, 2022-05-20:
This is a deep dive into my proposed rendering API and architectural changes with an explanation of problems, solutions, and impact to existing API. It is inclusive of these two other issues I filed:

- https://github.com/WildernessLabs/Meadow.Foundation/issues/333
- https://github.com/WildernessLabs/Meadow.Foundation/issues/336

**Context:**
I wrote the [Glade2d renderer](https://github.com/profexorgeek/Glade2d) as a light game engine for the Meadow, using the MeadowGraphics. While working on the 2d renderer, I found some friction with the MeadowGraphics API and had some early ideas for improvements, which I shared with @adrianstevens. My plan was to do some light refactoring to make the API a bit more flexible, while minimizing breakage.

When I set out to do this, I felt a bit like I was swimming upstream but couldn't put my finger on why. I had a collaborative jam session with my coworker @pwiest where we reviewed the entire render stack from top to bottom to see what options we had for optimization and came up with a desired architectural pattern that should probably be the direction Meadow takes with regards to sending content to pixel displays.

**Current:**

MeadowGraphics provides a drawing API and expects an `IGraphicsDisplay`. This interface is what all display devices implement to provide a driver. There is also an `IDisplayBuffer` interface, which the `IGraphicsDisplay` keeps. This buffer represents what should be drawn to the display and when `Show` is called on the `IGraphicsDisplay`, it sends the byte data across the bus to update the display. 

The similarity of the `IDisplayBuffer` and `IGraphicsBuffer` is confusing and also doesn't represent what's actually happening at the hardware level. For example, you can call `IGraphicsDisplay.DrawPixel`, which really just calls `IDisplayBuffer.SetPixel` on its buffer.

The architectural problem is that the separation between the concept of a buffer of texture data, a renderer that can perform pixel-level transforms, and a driver that streams bits to a device, is muddy. There is opportunity to do a clean separation here that will scale as you add device types and as you light up hardware-level acceleration.

**Ideal:**

Traditionally, graphics programmers have the concept of a graphics card they can program against (which is usually wrapped with a rendering library like DirectX). The graphics card handles primitive drawing operations as well as sending that data to the monitor, which is plugged directly into the card.

However, the Meadow chip doesn't quite work like this. It doesn't have discrete memory or graphics outputs and it doesn't [yet] have hardware that provides primative ops. It should separate the concept of the monitor driver, from the graphics renderer, from the pixel data buffer.

Existing capabilities can be refactored to better distinguish between these functions and allow work to eventually move from "software to hardware" layer when/if new capabilities are lit up on the chip. Meaning if hardware is capable of manipulating buffers, the `MeadowGraphics` class should stop doing that in .NET code and pass the buffers to the underlying hardware layer to transform.

The specific implementation should work as follows...

**_Monitor Driver_**
Right now the drivers that inherit from TftSpiBase are essentially a _monitor_ driver. They take a buffer full of bits and stream it across some wires to a display. But the current implementation acts like a cross between a graphics card and a monitor link. They provide methods to set pixels or fill the buffer. This is confusing because similar methods already exist on the buffer and the naming is different. But the behavior is the same, bytes in a buffer are manipulated. Additionally, some of these hardware driver calls may do pixel conversions on the fly, which hides performance problems (frequent bit depth conversions at rapid refresh rates when the buffer doesn't match the display). Display drivers should _only_ be responsible for initializing the hardware and sending data to the hardware. Their interface should be something like this:

```csharp
public interface IDisplayDriver
{
    // share key capabilities of the display hardware
    ColorMode ColorMode {get;}
    int BitDepth {get;}
    int Width {get;}
    int Height {get;}

    // static factory method provides a buffer that is the right size and depth for this display device
    static IPixelBuffer GetCompatibleBuffer();

    // pass the provided buffer to the display device, this should throw errors
    // if the buffer is not the right depth and size
    static ShowBuffer(IPixelBuffer buffer);

    // method to set specific color mode in case the device
    // can support multiple color modes
    bool TrySetColorMode(ColorMode targetMode);

   // note there are NO methods defined here for setting pixel data
   // because that happens at the buffer level and is not the responsibility
   // of this interface!
}
```

**_Texture Buffer_**
Right now, the uses of `Buffer` are confusing. First, `System.Buffer` is an existing class name (and `Buffer.BlockCopy` should probably be used instead of `Array.Copy` for performance reasons). This will really make the semantic usage of buffer confusing. In all cases with MeadowGraphics, buffer actually means 2d pixel data. We can use this to differentiate the intent of the interface.

This interface and its implementing classes would be responsible for maintaining a buffer of pixel data at a specific bit depth with capabilities to blit pixel data between buffers. This has minimal changes from IDisplayBuffer.

```csharp
public interface IPixelBuffer
{
    // share key attributes of this buffer
    ColorMode ColorMode {get;}
    int BitDepth {get;}
    int Width {get;}
    int Height {get;}
    int ByteCount {get;}
    byte[] Buffer {get;}

    // These methods already exist on IDisplayBuffer
    void Clear();
    void Fill(Color color);
    void Fill(Color color, int x, int y, int width, int height);
    void SetPixel(int x, int y, Color color);
    Color GetPixel(int x, int y);

    // This could "magically" convert the incoming buffer by
    // checking its bit depth and performing depth conversions from
    // source to target while also handling scaling
   void CopyFrom(IPixelBuffer newBuffer, int scale = 1, bool ignoreOutOfBounds = true);
}
```

**_Graphics Renderer_**
The graphics renderer would essentially be the `MeadowGraphics` class in its exact form today! There should be minimal API changes except that this class should be designed to extend with more robust drawing capabilities. Additionally, this class should
not operate on a Display Device because a Display Device no longer has its own buffer. Instead, this would get a buffer using the `IDisplayDriver`'s `GetCompatibleBuffer` method and the implementing dev could choose how many buffers they are compositing and working with before they actually send a final buffer to the display driver.

When/if dedicated hardware for faster buffer management is made available, this class could wrap the new methods without breaking drivers or buffers.

**Existing API Impact**

These changes will have the following impact to the API:

- IDisplayBuffer will be renamed to IPixelBuffer and one method added to the interface
- Existing implementations of IDisplayBuffer will have some new properties to implement and may choose to throw Not Implemented Exceptions for `CopyFrom` conversions they don't support
- IGraphicsDisplay` will be renamed to `IDisplayDriver` and many methods removed. Drivers will be updated to fit the interface. Drivers will no longer hold their own buffer and will throw exceptions when asked to `ShowBuffer` for an `IPixelBuffer` that doesn't match their supported buffer.
- The `MeadowGraphics` class will be refactored to be more flexible. It will maintain a buffer for the `IDisplayDriver` but will offer advanced implementations to do more flexible buffer management.
- Implementing tutorials that only call MeadowGraphics methods should not break.
- Some tutorials may need minor interface renames.
- Tutorials that directly interact with `IDisplayBuffer` or `IGraphicsDevice` will break.

Please let me know your thoughts or concerns. I can implement this refactor on my fork and PR it against main referencing this issue. I do not plan to update tutorials that this may break, I would leave that up to those maintainers.

adrianstevens, 2022-05-20:
This looks fantastic - I'll give it a full review in an hour but I think this is a big improvement

profexorgeek, 2022-05-21:
Leaving update notes here for myself because I'm at a stopping place for the night and in case anyone is interested in seeing progress. I'm working on this on this branch:
https://github.com/profexorgeek/Meadow.Foundation/tree/graphics-api

- I was able to convert `IDisplayBuffer` to `IPixelBuffer`, update all implementations with a safer API that doesn't use method hiding, allows better catching of slow buffer copies, and provides useful helper properties for bit depth, color mode, and byte array size
- I wrote a [render test in Glade2d that tests raw rendering speeds](https://github.com/profexorgeek/Glade2d/blob/master/Glade2d/MicroGraphicsTest.cs) and ran it before/after the update. Render speeds were the same for everything my device supports so there don't seem to be perf regressions.
- This API starts to bridge the gap so `IGraphicsDisplay` and `IPixelBuffer` have clearly differentiated roles
- The idea of the display driver not having a buffer might be bad. Now that I examine things outside of `MeadowGraphics` there are lots of driver implementations outside of Spi displays that rightfully expect the driver to maintain a buffer
- While the driver maybe _should_ have a buffer, it should still probably not have it's own draw calls that confusingly overlap with buffer draw calls. These can be removed and re-routed directly to the buffer owned by the driver?
- 3/6 tests are failing. I suspect these are easy fixes and 2 may have already been failing. I need to upgrade from VS Community to actually view the tests I think 🤔
- When this is all done, I still need to update `MeadowGraphics` to accomplish the original goal of supporting double/multi buffering

profexorgeek, 2022-05-21:
@adrianstevens I inspected the failing unit tests and they _shouldn't_ be failing because of things I did. So, I swapped branches to `main` and ran the unit tests again... they failed on main!

There are two important things here: first, there is something wrong with color conversion for Rgb565. I added some asserts and stepped through and the numbers aren't as expected.

However, I can't wade any deeper than I am on this particular PR effort so I'm going to leave these tests failing, since my work doesn't seem to be what broke them, and revisit that when this refactor is done.

closed 2022-05-26 by adrianstevens