WildernessLabs/Meadow.Foundation #344: PushButton loses PressEnded when clicking the button very fast

HakanL, 2022-06-06:
I have an application that subscribes to both PressStarted and PressEnded. I log to the console in the first line of each method. If I click the button super-fast then I only receive the PressStarted event, PressEnded is never invoked. Pressing the button normally works fine.

HakanL, 2022-06-06:
It looks like the issue is in this piece of code, I assume the elapsed time is 0 (immeasurable):
https://github.com/WildernessLabs/Meadow.Foundation/blob/e532eef01b71206038a0e8c4edfb6bf7afc7ef36/Source/Meadow.Foundation.Core/Sensors/Buttons/PushButton.cs#L195

adrianstevens, 2022-07-12:
Thanks @HakanL - I've tested and it looks like we're losing the event. I think we can remove that check regardless (I don't believe it buys us anything) but I don't think this will be fixed until we have JIT support (i.e. I think we're losing the event due to performance)

HakanL, 2022-07-12:
Thanks @adrianstevens, I'm not sure if we're losing the event due to performance, I switched to use a direct digital input and I don't get any lost events.

adrianstevens, 2022-11-17:
We should retest with JIT enabled on RC1

adrianstevens, 2023-01-25:
I haven't been able to reproduce this behavior on new versions of the OS - closing issue but don't hesitate to reopen

closed 2023-01-25 by adrianstevens