WildernessLabs/Meadow.Foundation #438: Tcs3472x Driver

asharzaman, 2022-10-26:
I am trying to use the Tcs34725 sensor with F7v1 and everything is wired up correctly. The application does detect the correct sensor. It doesn't matter what settings I try, the sensor is not detecting any changes. All of the RGB and Lux values are always zero. I have tried using it outside, inside, using a flashlight, or covering the sensor with my hand. 

I am using the example from: https://github.com/WildernessLabs/Meadow.Foundation/blob/e532eef01b71206038a0e8c4edfb6bf7afc7ef36/Source/Meadow.Foundation.Peripherals/Sensors.Light.Tcs3472x/Samples/Tcs3472x_Sample/MeadowApp.cs

Thanks

adrianstevens, 2022-11-17:
Unfortunately, this driver isn't complete and was accidentally published. I'll open a new driver request issue.

adrianstevens, 2022-11-17:
opened issue #479

adrianstevens, 2023-07-11:
closing issue as it's replaced by #479

closed 2023-07-11 by adrianstevens