WildernessLabs/Meadow.Foundation #504: Driver request: NeoPixel FeatherWing - 4x8 RGB LED Add-on

kingdona, 2022-12-08:
Willing to help where I can.

https://www.adafruit.com/product/2945?gclid=CjwKCAiAs8acBhA1EiwAgRFdw4ZAunUq0VBeOxXWrehgz2fSmY4xtclaX_PrAR4ht5mEj-6Xzh6aUxoCOMUQAvD_BwE

adrianstevens, 2022-12-09:
@kingdona unfortunately we don't yet have support for the 1-wire bus protocol we need to support NeoPixels but it's on the roadmap. It'll likely land shortly after Meadow hits v1.0.

I have one of these Featherwings and I'd love to see it working :)

@ctacke for input and visibility

kingdona, 2022-12-09:
@adrianstevens thanks for the update and information!

UeberDaniel, 2023-07-04:
Created a driver for it, would be good if it would be added to the Meadow.Fundation repository.
See https://github.com/UeberDaniel/Ws2812Display

kingdona, 2023-07-06:
@adrianstevens I received notification that a driver had been created for this by @UeberDaniel but didn't see the message in this thread. I'm a little confused if this applies to the featherwing I have or not that was linked in my original message for this request.

UeberDaniel, 2023-07-06:
@kingdona There is a partial driver. Unfortunately, it will not work properly for the layout of this panel. It would only work as a 4x8 matrix, i.e. 4 pixels on the x-axis and 8 pixels on the y-axis. The display would also be mirrored. When I noticed this, I deleted the comment shortly afterwards.
Here is the driver: https://github.com/UeberDaniel/Meadow-Ws2812Display-Driver

adrianstevens, 2023-08-14:
This driver has now been added and will ship with the next release

closed 2023-08-29 by adrianstevens