WildernessLabs/Meadow.Foundation #575: Read Method on SCD41 Blocks Forever

thegreatco, 2023-03-19:
**Describe the bug**
When setting up the SCD41 and attempting to call the `Read` method, calls appear to block forever.

**To Reproduce**
Steps to reproduce the behavior:
1. Initialize an SCD41
2. Call `.Read()` and `await` the result

**Expected behavior**
Method returns after a reasonable amount of time.

**Meadow.Foundation (please complete the following information as best as you can):**
 - `Meadow.Foundation.Sensors.Environmental.Scd4x` : 0.95.0

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11
 - IDE and version: Visual Studio 17.6.0 Preview 2.0
 - Meadow extension for IDE version: VS 2022 0.95.0

**Meadow (please complete the following information as best as you can):**
```
Board Information - Model: F7Micro, Hardware version: F7FeatherV2, Device name: MeadowEnvSense
Hardware Information - Processor type: STM32F777IIK6, ID: 38-00-39-00-08-51-38-34-35-31-38-33, Serial number: 337131773438, Coprocessor type: ESP32
Firmware Versions - OS: 0.9.4.0 Mono: 0.9.4.0, Coprocessor: 0.9.4.0
```

**Additional context**
Calling `StartUpdating()` before calling `Read()` causes `Read` to stop blocking forever.

adrianstevens, 2023-03-30:
thx @thegreatco - I'll take a look

adrianstevens, 2023-07-11:
#677 duplicates and superceeds this issue - closing