WildernessLabs/Meadow.Foundation #628: Cannot build Sensors.Hid.Keyboard

patridge, 2023-04-23:
Trying to build the Sensors.Hid.Keyboard project results in five errors like this.

> The type or namespace name 'NativeException' could not be found (are you missing a using directive or an assembly reference?)

From lines like this:
https://github.com/WildernessLabs/Meadow.Foundation/blob/main/Source/Meadow.Foundation.Peripherals/Sensors.Hid.Keyboard/Driver/Keyboard.cs#L198

I'm trying to build this to be able to get the Meadow.Windows.Samples solution to build.

It sounds like this driver isn't part of the Meadow.Foundation.sln because it is Windows-specific, which might be another problem. Is it just Windows or can it be built for other desktop OS platforms? If it's just Windows, can the name be something like `WindowsKeyboard` instead?

**Meadow.Foundation (please complete the following information as best as you can):**
 - Latest `develop` branch source

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11
 - IDE and version: Visual Studio Professional 2022 v17.5.3
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v0.96.2

patridge, 2023-04-23:
Looks like this project builds if you add a reference to Meadow.Core, but that isn't acceptable within the current Meadow.Foundation plans.

```
<ProjectReference Include="..\..\..\..\..\Meadow.Core\source\Meadow.Core\Meadow.Core.csproj" />
```

adrianstevens, 2023-04-23:
Ah ok - definitely something we need to solve

adrianstevens, 2023-07-11:
I believe this is fixed - closing

closed 2023-07-11 by adrianstevens