WildernessLabs/Meadow.Foundation #631: MCP23xxx.cs GP7 error in Meadow.Foundation v0.96

kaarew, 2023-04-24:
I'm not able to set up GP7 as an input pin. (GP0-GP6 works fine)

Steps to reproduce the behavior:
1. Set up the MCP23008 on a breadboard as shown in the hackster sample.
2. Use the following code to test:
            var i2cBus = Device.CreateI2cBus();
            mcp = new Mcp23008(i2cBus, 39);  // 39 not very intuitive. Ref. comment further down.
            mcp.CreateDigitalInputPort(mcp.Pins.GP6);   // Works fine.
            mcp.CreateDigitalInputPort(mcp.Pins.GP7);   // DOES NOT WORK.

Error message:
App Error in App Initialize: Specified argument was out of the range of valid values.

Possible solution in Mcp23xxx.cs:
In method ClearRegisterBit(byte register, int bit) this line:
            if (bit >= 7 || bit < 0) { throw new ArgumentOutOfRangeException(); }
...should be changed to:
            if (bit > 7 || bit < 0) { throw new ArgumentOutOfRangeException(); }
Same code can also be found in SetRegisterBit(byte register, int bit), and should probably be changed too.

Note also that in Mcp23008.cs, this constructor would be a nice addition:
        public Mcp23008(II2cBus i2cBus, bool A2, bool A1, bool A0, IDigitalInputPort interruptPort = null, IDigitalOutputPort resetPort = null) :
            base(i2cBus, (byte)(32 + (A2 ? 4 : 0) + (A1 ? 2 : 0) + (A0 ? 1 : 0)), interruptPort, resetPort)
        { }
Ref: https://github.com/WildernessLabs/Meadow.Core.Samples/issues/63

**Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV2
 - Meadow OS version: v0.9.6.3
 - If they are different, please provide these versions as well.
   - Meadow Mono version: v0.9.6.3
   - Meadow coprocessor version: v0.9.6.0

adrianstevens, 2023-04-24:
thx @kaarew - I'll get that fixed right away

adrianstevens, 2023-04-24:
Same bug is in SetRegisterBit

commit 9650c96d (2023-04-24): Fix bounds checking in SetRegsiterBit and ClearRegisterBit - fixes #631
closed 2023-05-06 by adrianstevens