WildernessLabs/Meadow.Foundation #704: MCP23S17 Example Throws Null Reference Exception - Possible Issue with Class or Documentation Errors

asieradzk, 2023-07-13:
Hello Meadow Foundation team,

I am currently experiencing an issue with the MCP23S17 example that I believe others may have encountered. While testing the MCP23S17, I keep encountering a null reference exception whenever I try to access an instance of mcp. I've thoroughly reviewed my wiring multiple times and it seems to be correct, leading me to believe that the problem might be elsewhere.

I am considering two possibilities: a mistake I might be making that I'm not aware of or a potential issue with the MCP23S17 class itself. I also noticed that the XML comments appear to be not working correctly and there seem to be errors in the official documentation, particularly in the wiring guide.

Any insight or assistance regarding this matter would be greatly appreciated. Thanks in advance for your support.
I am also quite new to meadow so not sure how to debug this myself.

Meadow os version 1.0.2 and 1.1

adrianstevens, 2023-09-06:
Hi @asieradzk the null reference exception is most likely in the code vs a wiring issue

adrianstevens, 2023-09-06:
and let's consolidate the discussion in #753  - closing issue

closed 2023-09-06 by adrianstevens