WildernessLabs/Meadow.Foundation #722: RgbLedColors does not support Black

ctacke, 2023-08-10:
**Describe the bug**
Turning RgbLed off is not consistent with API

Setting the color of the RgbLed is done with `RgbLedColors` but if you want "off" you have to use `IsOn = false` which is not intuitive nor consistent with our other LED APIs.

adrianstevens, 2023-08-14:
We should talk it through. Not supporting black was an intentional design decision to align with the single color led api.

The issue with black is, what does it mean to set IsOn to true if the led is black? We could switch it to white?

jorgedevs, 2023-08-14:
Not sure I follow about turning off an RgbLed is not consistent with the rest of LED drivers.
 
 To turn off an ILed:
  - Use the `IsOn` property for Led (sets digital output state to false)
  - Use the `IsOn` property for PwmLed (sets duty cycle to 0)
  - Use the `IsOn` property for RgbPwmLed  (sets duty cycle to 0)
  
 Thats why we removed setting the color to black to turn off the RgbLed, use the `IsOn` property like the other Led drivers.

jorgedevs, 2023-08-15:
Closing this issue, as I discussed with Chris about our design decision.