WildernessLabs/Meadow.Foundation #730: MicroLayout - DisplayLabel should clip text if its outside of width and heigth values

jorgedevs, 2023-08-15:
On a Project Lab v3, conside the following MicroLayoutSample

```csharp
_screen.Controls.Add(
                new DisplayBox(0, 0, _screen.Width, _screen.Height)
                {
                    ForeColor = Color.White
                },
                new DisplayLabel(60, 20, 100, 40)
                {
                    Text = "Welcome to MicroLayout!",
                    TextColor = Color.Black,
                    BackColor = Color.Red,
                    Font = new Font12x20(),
                    VerticalAlignment = VerticalAlignment.Center,
                    HorizontalAlignment = HorizontalAlignment.Center
                });
```

Notice that the background color (red) delimits the bounds of this control, yet on the Text is drawn outside these boundaries. 

[image]

ctacke, 2023-08-16:
@adrianstevens any thoughts on how to achieve this easily?  uG doesn't have any sort of clipping region - we just draw text at a location.  I don't see any sort of "width" or "height" for the drawing bounds, so I assume it will just flow to the edge of the screen.

adrianstevens, 2023-08-29:
One solve is to write to a temporary buffer that matches the background size which will handle the crop, and then blit that back to the main buffer.

It also shouldn't be too hard to extend the APIs to specify a drawing region - I'll take a look.