WildernessLabs/Meadow.Foundation #731: MicroLayout - Add background color to DisplayButton

jorgedevs, 2023-08-15:
To further customize buttons like in MAUI or Xamarin.Forms, would be nice to have the ability to set the background Color like in DisplayLabel.

Its nice to hace ShadowColor, tho.

```csharp
new DisplayButton(170, 178, 120, 40)
{
    Text = "Button 2",
    TextColor = Color.White,
    ShadowColor = Color.FromHex("#333333"),
    Font = new Font12x20(),
}
```

ctacke, 2023-08-16:
Not sure where this color would reside...the button takes up the entire bounds of the control.  Do you mean the `ForeColor` property?  That's the color of the face of the button when unpressed - maybe we change it to "UnpressedColor"?

jorgedevs, 2023-08-16:
Thats the properpty i was lookin for. I can close this issue.

closed 2023-08-16 by jorgedevs