WildernessLabs/Meadow.Foundation #745: TaskCancelledException thrown when multiple Tasks call RgbPwmLed.StartPulse(color, duration) method.

doingnz, 2023-08-22:
**Describe the bug**
TaskCancelledException thrown when multiple Tasks call RgbPwmLed.StartPulse(color, duration) method. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create Project New Blinky
2. Add Tasks to task Run() that set the RGB LED colour by calling ShowColourPulse(...);
3. Enable Trace output.
4. Debug the application. 

**Expected behavior**
TaskCancelledException is not thrown. The App is not terminated. 

**Screenshots**
Example Task Run()
```
        public override Task Run()
        {
            Resolver.Log.Info("Run...");

            Task.Run(async () =>
            {
                while (true)
                {
                    await ShowColorPulse(Color.Red, defaultDuration / 2);
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    await ShowColorPulse(Color.Blue, defaultDuration / 2);
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    await ShowColorPulse(Color.Green, defaultDuration / 2);
                }
            });
```

**Developer tools :**
- Microsoft Windows [Version 10.0.22621.1992]
- Visual Studio Enterprise 2022 [Version v17.6.5]
- Meadow extension for IDE version: [VS 2022 Tools for Meadow v1.2.0]

**Meadow:**
```
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 22-00-26-00-11-51-36-30-33-33-37-33
    Serial number: 335D335F3036
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F4:0C

Firmware Versions
    OS: 1.2.0.1
    Mono: 1.2.0.1
    Coprocessor: 1.2.0.0
    Protocol: 7
```

**Additional context**
Additional problems exist in the current implementation:
- Both Task and CancellationTokenSource objects implement IDisposable, but the `StopAnimation` is not calling `Dispose()` method on either of these prior to setting them to null. Adding calls to `Dispose()` expose a race condition where `animationTask` can be cancelled and the `await animationTask` returns execution to this method, but this status is still propagating through the Tasks/objects, when `animationTask.Dispose()` is called, which throws an exception as it is unhappy to `Dispose()` a Task that is not yet completed execution! In the code below, a while loop is added to check animatedTask.IsCompleted to avoid this exception.  The code now runs for longer before it throws other exceptions. 
- There is a memory leak which can be revealed by logging GC usage.
- The value of Task.CurrentId is nearly always 0?

Sample code for StopAnimation
```
        public async Task StopAnimation()
        {
            LogMemory("StopAnimation", $"{animationTask != null}", Task.CurrentId ?? 0, comment:"Start"); //, count = 0, string comment = "")

            if (animationTask != null)
            {
                cancellationTokenSource?.Cancel();

                try
                {
                    await animationTask;
                    Resolver.Log.Info($"StopAnimation: Status {animationTask.Status}. After await");
                    while (!animationTask.IsCompleted)
                    {
                        Resolver.Log.Info($"StopAnimation: Status {animationTask.Status}. Waiting for IsCompleted");
                        Thread.Sleep(100);
                    }
                    cancellationTokenSource.Dispose();
                    cancellationTokenSource = null;
                    animationTask.Dispose();
                }
                catch (Exception e)
                {
                    Resolver.Log.Info($"StopAnimation: Status {animationTask.Status}. Exception.Message={e.Message}");
                }
                finally
                {
                    animationTask = null;
                    LogMemory("StopAnimation", $"{animationTask != null}", Task.CurrentId??0, comment: "End"); //, count = 0, string comment = "")
                }
            }
        }

        public static string LogMemory(string function, string commentCount = "", int count = 0, string comment = "")
        {
            GC.Collect();
            GC.WaitForPendingFinalizers();
            string msg = $"LogMemory, {TimeStamp(),25},{GC.GetTotalMemory(true),12},\"{function,-30}\",\"{(string.IsNullOrWhiteSpace(commentCount) ? "" : commentCount)}\",{(string.IsNullOrWhiteSpace(commentCount) ? "" : count.ToString())},\"{(string.IsNullOrWhiteSpace(comment) ? "" : comment)}\"";
            Resolver.Log.Info(msg);
            return msg;
        }

        public static string TimeStamp()
        {
            return $"{System.DateTime.Now:yyyy-MM-ddTHH:mm:ss}";
        }
```

doingnz, 2023-09-18:
Please don't overlook this issue as it raises questions and potential issues or race-conditions in the underpinning handling of tasks and cancellation tokens, which have broad implications to Meadow.OS. 

There is a bug/design problem in handling of cancellations for RGB LED, but that is not as important :-).

Please let me know if the previous explanation is not clear about the task/cancellation problems.

adrianstevens, 2023-09-27:
I've been testing this week and I haven't yet seen a TaskCancelledException but it is freezing when debugging

adrianstevens, 2023-09-27:
@CartBlanche here's a case where deploying is working but debubbing fails

doingnz, 2023-09-27:
With 1.3.4.0, If I make a new Blinky Project and just add the 3xTasks to set LED colors, then I get a TaskCancelledException in 2 or 3 seconds. 
```App Error in App Run: A task was canceled.
System.Threading.Tasks.TaskCanceledException: A task was canceled.
at Meadow.Foundation.Leds.RgbPwmLed.StopAnimation () <0xc1357f18 + 0x000d8> in <d663bcbce379408896530e841a6940d0>:0
at Meadow.Foundation.Leds.RgbPwmLed.StartPulse (Meadow.Foundation.Color color, System.TimeSpan pulseDuration, System.Single highBrightness, System.Single lowBrightness) <0xc1363528 + 0x000d4> in <d663bcbce379408896530e841a6940d0>:0
at MeadowBlinkyTest.MeadowA
p.ShowColorPulse (Meadow.Foundation.Color color, System.TimeSpan duration) [0x00054] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:112
at MeadowBlinkyTest.MeadowApp.CycleColors (System.TimeSpan duration) [0x0011e] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:96
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) <0xc0bf20c0 + 0x002
8> in <d4f614572c6047c08610bc5349864be2>:0
App shutting down
Shutdown```

doingnz, 2023-09-27:
Debugging works for me, but you need to set the breakpoint in the `async Task ShowColorPulse(Color color, TimeSpan duration)` method to avoid the app being shutdown due to the TaskCancelledException.

doingnz, 2023-09-27:
with 1.3.4.0, the following change to Blinky and Trace logging enabled:
```
        public override Task Run()
        {
            Resolver.Log.Info("Run...");

            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(100);
                    Resolver.Log.Info("Red LED");
                    await ShowColorPulse(Color.Red, defaultDuration / 2);
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(100);
                    Resolver.Log.Info("Blue LED");
                    await ShowColorPulse(Color.Blue, defaultDuration / 2);
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(100);
                    Resolver.Log.Info("Green LED");
                    await ShowColorPulse(Color.Green, defaultDuration / 2);
                }
            });

            return CycleColors(TimeSpan.FromMilliseconds(1000));
        }
```
it appears to fail every time after Log Trace of `Green LED` as follows. Tried a couple of times.
```
Initializing App
Initialize...
Running App
Run...
Cycle colors...
Red LED
Blue LED
Green LED
App Error in App Run: A task was canceled.
System.Threading.Tasks.TaskCanceledException: A task was canceled.
...
```

doingnz, 2023-09-27:
if you change StopAnimation as suggested, then the code will sort of run for hours before it stops. The failure mode may have changed subtly with 1.3.4.0 from what I remember with previous release. With 1.3.4.0 it's as if the Task schedular stacks up calls to one or two Tasks and they get called repeatedly for ages before calling one of the others. e.g. currently only RED colour pulsing and no BLUE or GREEN and adding breakpoint in the "RED" task is being hit, and none in the other colors. A short time ago it was blue and green and no red.

doingnz, 2023-10-02:
@adrianstevens have you tried setting a break-point in `async Task ShowColorPulse(Color color, TimeSpan duration)` prior to starting a debugging session to confirm this test app does not prevent debugging in Visual Studio?

doingnz, 2023-10-02:
If you include the `Resolver.Log.Info(...)` and `await Task.Delay(n)` where n is 1 or higher, then it will run for short time till the main Task running `CycleColours()` ends with the following exception
```
System.NullReferenceException: Object reference not set to an instance of an object.
at Meadow.Foundation.Leds.RgbPwmLed.StopAnimation () [0x001af] in D:\Meadow\Meadow.Foundation\Source\Meadow.Foundation.Core\Leds\RgbPwmLed.Animations.cs:44
at Meadow.Foundation.Leds.RgbPwmLed.StartPulse (Meadow.Foundation.Color color, System.TimeSpan pulseDuration, System.Single highBrightness, System.Single lowBrightness) [0x00043] in D:\Meadow\Meadow.Foundation\Source\Meado
.Foundation.Core\Leds\RgbPwmLed.Animations.cs:199
at MeadowBlinkyTest.MeadowApp.ShowColorPulse (Meadow.Foundation.Color color, System.TimeSpan duration) [0x00054] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:120
at MeadowBlinkyTest.MeadowApp.CycleColors (System.TimeSpan duration) [0x000bd] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:112
at
Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x00180] in D:\Meadow\Meadow.Core\source\Meadow.Core\MeadowOS.cs:103
App shutting down
Shutdown
```

@adrianstevens Please also check all three Tasks (RED, GREEN, BLUE) are running as typically one of them runs once and is never scheduled again!

adrianstevens, 2023-10-03:
The color of the led flashing seems to be random, I have had it flash all 3, but typically I see one or two

adrianstevens, 2023-10-03:
I set mine to 100s (so it'll be 50 with the / 2)

```
Initialize...
Run...
Cycle colors...
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
```

adrianstevens, 2023-10-03:
At 1s:

```
Initialize...
Run...
Cycle colors...
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Blue LED
Green LED
Red LED
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
Red LED
Red LED
Blue LED
Green LED
Red LED
Red LED
Blue LED
Green LED
Red LED
Blue LED
Green LED
```

adrianstevens, 2023-10-03:
here's my MeadowApp.cs 

```
using Meadow;
using Meadow.Devices;
using Meadow.Foundation;
using Meadow.Foundation.Leds;
using Meadow.Peripherals.Leds;
using System;
using System.Threading.Tasks;

namespace BlinkCrash
{
    // Change F7FeatherV2 to F7FeatherV1 for V1.x boards
    public class MeadowApp : App<F7FeatherV2>
    {
        RgbPwmLed onboardLed;

        readonly TimeSpan defaultDuration = TimeSpan.FromSeconds(5);

        public override Task Initialize()
        {
            Resolver.Log.Info("Initialize...");

            onboardLed = new RgbPwmLed(
                redPwmPin: Device.Pins.OnboardLedRed,
                greenPwmPin: Device.Pins.OnboardLedGreen,
                bluePwmPin: Device.Pins.OnboardLedBlue,
                CommonType.CommonAnode);

            return base.Initialize();
        }

        public override Task Run()
        {
            Resolver.Log.Info("Run...");

            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(100);
                    Resolver.Log.Info("Red LED");
                    await ShowColorPulse(Color.Red, defaultDuration / 2);
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(100);
                    Resolver.Log.Info("Blue LED");
                    await ShowColorPulse(Color.Blue, defaultDuration / 2);
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(100);
                    Resolver.Log.Info("Green LED");
                    await ShowColorPulse(Color.Green, defaultDuration / 2);
                }
            });

            return CycleColors(TimeSpan.FromMilliseconds(1000));
        }

        async Task CycleColors(TimeSpan duration)
        {
            Resolver.Log.Info("Cycle colors...");

            while (true)
            {
                await ShowColorPulse(Color.Blue, duration);
                await ShowColorPulse(Color.Cyan, duration);
                await ShowColorPulse(Color.Green, duration);
                await ShowColorPulse(Color.GreenYellow, duration);
                await ShowColorPulse(Color.Yellow, duration);
                await ShowColorPulse(Color.Orange, duration);
                await ShowColorPulse(Color.OrangeRed, duration);
                await ShowColorPulse(Color.Red, duration);
                await ShowColorPulse(Color.MediumVioletRed, duration);
                await ShowColorPulse(Color.Purple, duration);
                await ShowColorPulse(Color.Magenta, duration);
                await ShowColorPulse(Color.Pink, duration);
            }
        }

        async Task ShowColorPulse(Color color, TimeSpan duration)
        {
            await onboardLed.StartPulse(color, duration / 2);
            await Task.Delay(duration);
        }
    }
}
```

adrianstevens, 2023-10-03:
Ok after leaving it running I finally saw the error: 

```
Blue LED
App Error in App Run: A task was canceled.
System.Threading.Tasks.TaskCanceledException: A task was canceled.
Green LED
at Meadow.Foundation.Leds.RgbPwmLed.StopAnimation () [0x0007b] in <d663bcbce379408896530e841a6940d0>:0
at Meadow.Foundation.Leds.RgbPwmLed.StartPulse (Meadow.Foundation.Color color, System.TimeSpan pulseDuration, System.Single highBrightness, System.Single lowBrightness) [0x0007e] in <d663bcbce379408896530e841a6940d0>:0
at BlinkCrash.MeadowApp.ShowColorPulse (Meadow.Founda
ion.Color color, System.TimeSpan duration) [0x000a0] in <c151d0be07244ab79ecb32d45ee997f9>:0
at BlinkCrash.MeadowApp.CycleColors (System.TimeSpan duration) [0x005e8] in <c151d0be07244ab79ecb32d45ee997f9>:0
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x00164] in <d4f614572c6047c08610bc5349864be2>:0
Red LED
```

doingnz, 2023-10-03:
It can be very quick or take a short time. It will always fail with the same error. I don't think this exception should happen. Adding a try/catch in the StopAnimation to trap this error leads to other problems with Task scheduling.

doingnz, 2023-10-03:
The reason for the random colour changes is each task tries to cancel any previous Task that was animating the LED before it has done a single "pulse".  This is a stress test for the Task and Cancellation logic. There may be an "alternate' design philosophy that waits for at least one animation cycle before cancel, but that is a different design question on how the library should operate... The core problem from my perspective currently is the task cancellation is not working as expected and then other scheduling issues occur is one tries to trap and handle TaskCanceledException.

doingnz, 2023-10-03:
for interest, if the 3 tasks don't animate the RGB led, and only increment a counter, then it will run for hours. The main Task still cycles the colours. This helps confirm that it is the competing cancellation of the Tasks that leads to an internal race condition and the unexpected error,

```
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(10);
                    //Resolver.Log.Info("Blue LED");
                    Interlocked.Increment(ref countBlue);
                    //await Task.Delay(defaultDuration / 2);
                    //await ShowColorPulse(Color.Blue, defaultDuration / 2);
                    Thread.Yield();
                }
            });
```

Added Logging to ShowColorPulse() to output 3 counters and number of times  ShowColorPulse(0 called. Also memory allocated which confirms there is a low memory leak in this simple app too :-)

```
Meadow StdOut: LogMemory,       2023-10-03T21:44:22,      883008,"CycleColors                   ","Count: Red=1455797271 Blue=1455776124 Green=1455753973",64339,""
```
if I restart the test app we can see the starting memory allocation is lower
```
Meadow StdOut: LogMemory,       2023-10-03T21:51:26,      605256,"CycleColors                   ","Count: Red=    9047 Blue=    9045 Green=    9039",1,""
```

adrianstevens, 2023-10-03:
Interesting that you're seeing memory go up with the code above - I expect to see a slow increase calling ShowColorPulse since it's creating a cancellation token and such.

Ideally the garbage collector kicks in at some point and cleans up. Running with local references for me seems stable but I see a steady increase in memory usage. I'll leave it running now, I'm curious to see if it collects.

adrianstevens, 2023-10-03:
And I see strange jumps in memory usage

```
Blue  led: 576,336 bytes
Green led: 576,536 bytes
Red   led: 576,736 bytes
Blue  led: 576,936 bytes
Green led: 577,136 bytes
Red   led: 653,664 bytes
Blue  led: 653,864 bytes
Green led: 654,064 bytes
Red   led: 654,264 bytes
Blue  led: 654,464 bytes
Green led: 654,664 bytes
```

```
Blue  led: 671,264 bytes
Green led: 671,464 bytes
Red   led: 671,664 bytes
Blue  led: 713,272 bytes
Green led: 713,472 bytes
Red   led: 713,672 bytes
Blue  led: 713,872 bytes
Green led: 714,072 bytes
```

```
Blue  led: 730,072 bytes
Green led: 730,272 bytes
Red   led: 730,472 bytes
Blue  led: 730,672 bytes
Green led: 730,872 bytes
Red   led: 771,968 bytes
Blue  led: 772,168 bytes
Green led: 772,368 bytes
Red   led: 772,568 bytes
Blue  led: 772,768 bytes
Green led: 772,968 bytes
```

adrianstevens, 2023-10-03:
@alexischr would this allocating more managed memory? and in which case, we'd expect to see a collection happen and _should_ be able to free up memory.

doingnz, 2023-10-03:
the test app uses the following that forces a GC at each Log.

```
        async Task ShowColorPulse(Color color, TimeSpan duration)
        {
            Interlocked.Increment(ref countCycleColors);
            await onboardLed.StartPulse(color, duration / 2);
            await Task.Delay(duration);
            //Resolver.Log.Info($"Count: Red={countRed} Red={countBlue} Red={countGreen} CycleColors={countCycleColors}");
            LogMemory("CycleColors", $"Count: Red={countRed,8} Blue={countBlue,8} Green={countGreen,8}", countCycleColors);
            Thread.Yield();
        }

        public static string LogMemory(string function, string commentCount = "", int count = 0, string comment = "")
        {
            GC.Collect();
            GC.WaitForPendingFinalizers();
            string msg = $"LogMemory, {TimeStamp(),25},{GC.GetTotalMemory(true),12},\"{function,-30}\",\"{(string.IsNullOrWhiteSpace(commentCount) ? "" : commentCount)}\",{(string.IsNullOrWhiteSpace(commentCount) ? "" : count.ToString())},\"{(string.IsNullOrWhiteSpace(comment) ? "" : comment)}\"";
            Resolver.Log.Info(msg);
            return msg;
        }

        public static string TimeStamp()
        {
            return $"{System.DateTime.Now:yyyy-MM-ddTHH:mm:ss}";
        }
```

adrianstevens, 2023-10-03:
roger that - let's move this back to the main Meadow Issues repo and the OS team looking at this

adrianstevens, 2023-10-06:
@doingnz the OS team has been making some improvements on time slicing and thread priorities. If you're feeling brave you can try our latest pre-release OS build:

`meadow download os -v 1.3.5.4`
`meadow flash os -v 1.3.5.4`

doingnz, 2023-10-06:
@adrianstevens v 1.3.5.4  has similar behaviour when the Meadow.Foundation nuget is referenced.
```
Cycle colors...
LogMemory,       2023-10-06T21:42:23,      646680,"CycleColors                   ","Count: Red=       1 Blue=       1 Green=       1",1,""
LogMemory,       2023-10-06T21:42:25,      650120,"CycleColors                   ","Count: Red=       1 Blue=       5 Green=       6",2,""
LogMemory,       2023-10-06T21:42:26,      653464,"CycleColors                   ","Count: Red=       1 Blue=       7 Green=       8",3,""
LogMemory,       2023-10-06T21:42:27,      656584,"CycleColors                   ","Count: Red=       1 Blue=       9 Green=      10",4,""
LogMemory,       2023-10-06T21:42:28,      659704,"CycleColors                   ","Count: Red=       1 Blue=      11 Green=      12",5,""
LogMemory,       2023-10-06T21:42:29,      662824,"CycleColors                   ","Count: Red=       1 Blue=      13 Green=      14",6,""
LogMemory,       2023-10-06T21:42:31,      665768,"CycleColors                   ","Count: Red=       1 Blue=      15 Green=      16",7,""
LogMemory,       2023-10-06T21:42:32,      669296,"CycleColors                   ","Count: Red=       1 Blue=      17 Green=      17",8,""
LogMemory,       2023-10-06T21:42:33,      672904,"CycleColors                   ","Count: Red=       1 Blue=      19 Green=      19",9,""
LogMemory,       2023-10-06T21:42:34,      676288,"CycleColors                   ","Count: Red=       1 Blue=      21 Green=      21",10,""
LogMemory,       2023-10-06T21:42:35,      676104,"CycleColors                   ","Count: Red=       1 Blue=      22 Green=      22",11,""
App Error in App Run: A task was canceled.
System.Threading.Tasks.TaskCanceledException: A task was canceled.
at Meadow.Foundation.Leds.RgbPwmLed.StopAnimation () <0xc134af58 + 0x000d8> in <d663bcbce379408896530e841a6940d0>:0
at Meadow.Foundation.Leds.RgbPwmLed.StartPulse (Meadow.Foundation.Color color, System.TimeSpan pulseDuration, System.Single highBrightness, System.Single lowBrightness) <0xc134a898 + 0x000d4> in <d663bcbce379408896530e841a6940d0>:0
at MeadowBlinkyTest.MeadowA
p.ShowColorPulse (Meadow.Foundation.Color color, System.TimeSpan duration) [0x00068] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:133
at MeadowBlinkyTest.MeadowApp.CycleColors (System.TimeSpan duration) [0x0059c] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:126
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) <0xc0bf2840 + 0x00
18> in <d4f614572c6047c08610bc5349864be2>:0
App shutting down
Shutdown
```
NOTE in above run "Red" Task is not being scheduled. The behaviour changes if you run multiple times. 

It can fail immediately, or after a few loops. 
```
Run...
Cycle colors...
LogMemory,       2023-10-06T22:02:08,      637272,"CycleColors                   ","Count: Red=       1 Blue=       1 Green=       0",1,""
App Error in App Run: A task was canceled.
System.Threading.Tasks.TaskCanceledException: A task was canceled.
at Meadow.Foundation.Leds.RgbPwmLed.StopAnimation () <0xc1343658 + 0x000d8> in <d663bcbce379408896530e841a6940d0>:0
at Meadow.Foundation.Leds.RgbPwmLed.StartPulse (Meadow.Foundation.Color color, System.TimeSpan pulseDuration, System.Single highBrightness, System.Single lowBrightness) <0xc1342f98 + 0x000d4> in <d663bcbce379408896530e841a6940d0>:0
at MeadowBlinkyTest.MeadowA
p.ShowColorPulse (Meadow.Foundation.Color color, System.TimeSpan duration) [0x00068] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:106
at MeadowBlinkyTest.MeadowApp.CycleColors (System.TimeSpan duration) [0x0011e] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:89
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) <0xc0bf2840 + 0x002
8> in <d4f614572c6047c08610bc5349864be2>:0
App shutting down
Shutdow
```
In some instances all the tasks are equally scheduled. 

Test application is as follows
```
    public class MeadowApp : App<F7FeatherV2>
    {
        RgbPwmLed onboardLed;

        private readonly TimeSpan defaultDuration = TimeSpan.FromMilliseconds(1000);

        public int countRed  = 0;
        public int countBlue = 0;
        public int countGreen = 0;
        public int countCycleColors = 0;

        public override Task Initialize()
        {
            Resolver.Log.Info("Initialize...");

            onboardLed = new RgbPwmLed(
                redPwmPin: Device.Pins.OnboardLedRed,
                greenPwmPin: Device.Pins.OnboardLedGreen,
                bluePwmPin: Device.Pins.OnboardLedBlue,
                CommonType.CommonAnode);
            
[... 73 lines cut ...]
            LogMemory("CycleColors", $"Count: Red={countRed,8} Blue={countBlue,8} Green={countGreen,8}", countCycleColors);
            //Thread.Yield();
        }

        async Task ShowColorPulseClassic(Color color, TimeSpan duration)
        {
            await onboardLed.StartPulse(color, duration / 2);
            await Task.Delay(duration);
        }

        public static string LogMemory(string function, string commentCount = "", int count = 0, string comment = "")
        {
            GC.Collect();
            GC.WaitForPendingFinalizers();
            string msg = $"LogMemory, {TimeStamp(),25},{GC.GetTotalMemory(true),12},\"{function,-30}\",\"{(string.IsNullOrWhiteSpace(commentCount) ? "" : commentCount)}\",{(string.IsNullOrWhiteSpace(commentCount) ? "" : count.ToString())},\"{(string.IsNullOrWhiteSpace(comment) ? "" : comment)}\"";
            Resolver.Log.Info(msg);
            return msg;
        }

        public static string TimeStamp()
        {
            return $"{System.DateTime.Now:yyyy-MM-ddTHH:mm:ss}";
        }
    }
```

doingnz, 2023-10-06:
An interesting example run where is ran for longer than normally before the Shutdown. 

Note how after the first few iterations the RED task is not getting a fair share of execution time as the count is going up a lot slower. 

If you look at the time steps on the left, there are some very large intervals. That is an indication the CycleColors(..) Task is not being scheduled. When the app is running, you can see this visually as the LogMemory lines are not displayed. 

Note the significant memory leak with the allocated memory doubling before the application is shutdown after the TaskCanceledException

```
Cycle colors...
LogMemory,       2023-10-06T22:05:05,      635536,"CycleColors                   ","Count: Red=       1 Blue=       1 Green=       1",1,""
LogMemory,       2023-10-06T22:05:07,      639496,"CycleColors                   ","Count: Red=       3 Blue=       3 Green=       4",2,""
LogMemory,       2023-10-06T22:05:09,      643488,"CycleColors                   ","Count: Red=       4 Blue=       5 Green=       6",3,""
LogMemory,       2023-10-06T22:05:10,      646608,"CycleColors                   ","Count: Red=       4 Blue=       7 Green=       8",4,""
LogMemory,       2023-10-06T22:05:11,      649728,"CycleColors                   ","Count: Red=       6 Blue=       9 Green=       8",5,""
LogMemory,       2023-10-06T22:05:12,      652824,"CycleColors                   ","Count: Red=       6 Blue=      11 Green=      10",6,""
LogMemory,       2023-10-06T22:05:13,      655944,"CycleColors                   ","Count: Red=       6 Blue=      13 Green=      12",7,""
LogMemory,       2023-10-06T22:05:14,      659064,"CycleColors                   ","Count: Red=       6 Blue=      15 Green=      14",8,""
LogMemory,       2023-10-06T22:05:15,      662184,"CycleColors                   ","Count: Red=       6 Blue=      17 Green=      16",9,""
LogMemory,       2023-10-06T22:05:17,      665328,"CycleColors                   ","Count: Red=       6 Blue=      19 Green=      18",10,""
LogMemory,       2023-10-06T22:05:18,      668448,"CycleColors                   ","Count: Red=       8 Blue=      21 Green=      18",11,""
LogMemory,       2023-10-06T22:05:19,      671544,"CycleColors                   ","Count: Red=       8 Blue=      23 Green=      20",12,""
LogMemory,       2023-10-06T22:05:20,      674688,"CycleColors                   ","Count: Red=       8 Blue=      25 Green=      22",13,""
LogMemory,       2023-10-06T22:05:21,      677784,"CycleColors                   ","Count: Red=       8 Blue=      27 Green=      24",14,""
LogMemory,       2023-10-06T22:05:22,      680904,"CycleColors                   ","Count: Red=       8 Blue=      29 Green=      26",15,""
LogMemory,       2023-10-06T22:05:23,      684024,"CycleColors                   ","Count: Red=       8 Blue=      31 Green=      28",16,""
LogMemory,       2023-10-06T22:05:24,      687144,"CycleColors                   ","Count: Red=       8 Blue=      33 Green=      30",17,""
LogMemory,       2023-10-06T22:05:25,      690264,"CycleColors                   ","Count: Red=       8 Blue=      35 Green=      32",18,""
LogMemory,       2023-10-06T22:05:26,      693384,"CycleColors                   ","Count: Red=       8 Blue=      37 Green=      34",19,""
LogMemory,       2023-10-06T22:05:27,      696528,"CycleColors                   ","Count: Red=       8 Blue=      39 Green=      36",20,""
LogMemory,       2023-10-06T22:05:29,      699648,"CycleColors                   ","Count: Red=      10 Blue=      41 Green=      36",21,""
LogMemory,       2023-10-06T22:05:30,      702744,"CycleColors                   ","Count: Red=      10 Blue=      43 Green=      38",22,""
LogMemory,       2023-10-06T22:05:31,      705888,"CycleColors                   ","Count: Red=      10 Blue=      45 Green=      40",23,""
LogMemory,       2023-10-06T22:05:32,      709008,"CycleColors                   ","Count: Red=      12 Blue=      47 Green=      40",24,""
LogMemory,       2023-10-06T22:05:33,      712128,"CycleColors                   ","Count: Red=      12 Blue=      49 Green=      42",25,""
LogMemory,       2023-10-06T22:05:34,      715224,"CycleColors                   ","Count: Red=      14 Blue=      51 Green=      42",26,""
LogMemory,       2023-10-06T22:05:35,      718368,"CycleColors                   ","Count: Red=      16 Blue=      53 Green=      42",27,""
LogMemory,       2023-10-06T22:05:36,      721464,"CycleColors                   ","Count: Red=      16 Blue=      55 Green=      44",28,""
LogMemory,       2023-10-06T22:05:37,      724608,"CycleColors                   ","Count: Red=      16 Blue=      57 Green=      46",29,""
LogMemory,       2023-10-06T22:05:38,      727728,"CycleColors                   ","Count: Red=      18 Blue=      59 Green=      46",30,""
LogMemory,       2023-10-06T22:05:39,      730824,"CycleColors                   ","Count: Red=      18 Blue=      61 Green=      48",31,""
LogMemory,       2023-10-06T22:05:41,      733944,"CycleColors                   ","Count: Red=      18 Blue=      63 Green=      50",32,""
LogMemory,       2023-10-06T22:05:42,      737064,"CycleColors                   ","Count: Red=      18 Blue=      65 Green=      52",33,""
LogMemory,       2023-10-06T22:05:43,      740184,"CycleColors                   ","Count: Red=      18 Blue=      67 Green=      54",34,""
[... cut ...] 896530e841a6940d0>:0
at MeadowBlinkyTest.MeadowA
p.ShowColorPulse (Meadow.Foundation.Color color, System.TimeSpan duration) [0x00068] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:106
at MeadowBlinkyTest.MeadowApp.CycleColors (System.TimeSpan duration) [0x0018f] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:90
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) <0xc0bf2840 + 0x002
8> in <d4f614572c6047c08610bc5349864be2>:0
App shutting down
Shutdown
```

doingnz, 2023-10-06:
if I comment out the the call to pulse the led from Red/Green/Blue Tasks and therefore avoid setting cancellation token...

```
        async Task ShowColorPulseClassic(Color color, TimeSpan duration)
        {
            //await onboardLed.StartPulse(color, duration / 2);
            await Task.Delay(duration);
        }
```
the code runs till memory is consumed by the leak.
```
Run...
Cycle colors...
LogMemory,       2023-10-06T22:17:29,      634752,"CycleColors                   ","Count: Red=       1 Blue=       1 Green=       1",1,""
LogMemory,       2023-10-06T22:17:31,      636080,"CycleColors                   ","Count: Red=       6 Blue=       6 Green=       5",2,""
LogMemory,       2023-10-06T22:17:32,      637280,"CycleColors                   ","Count: Red=       8 Blue=       8 Green=       7",3,""

... delete for clarity ....

LogMemory,       2023-10-06T23:35:57,     3051400,"CycleColors                   ","Count: Red=    9096 Blue=    9096 Green=    9096",3869,""
LogMemory,       2023-10-06T23:35:58,     3052024,"CycleColors                   ","Count: Red=    9099 Blue=    9098 Green=    9098",3870,""
LogMemory,       2023-10-06T23:35:59,     3052648,"CycleColors                   ","Count: Red=    9101 Blue=    9101 Green=    9101",3871,""
LogMemory,       2023-10-06T23:36:01,     3053272,"CycleColors                   ","Count: Red=    9104 Blue=    9103 Green=    9103",3872,""
LogMemory,       2023-10-06T23:36:02,     3054040,"CycleColors                   ","Count: Red=    9106 Blue=    9106 Green=    9106",3873,""
Could not allocate 1376 (1 * 1376) bytes
```

doingnz, 2023-11-10:
Testing with 1.5.0.3 is looking good.

I download 1.5.0.3 and flashed a feather V2. Deployed modified Blinky with the example code in the description to stress test the 4 tasks fighting to set the RGB LED.

Its been running correctly for nearly 4 hours without any errors and no memory leaks!
The counts are also balanced between red/green/blue showing these tasks are getting equal share compute time. 
```
Run...
Cycle colors...
LogMemory,       2023-11-09T03:34:32,      622288,"CycleColors                   ","Count: Red=       1 Blue=       0 Green=       1",2,""
LogMemory,       2023-11-09T03:34:33,      622480,"CycleColors                   ","Count: Red=       1 Blue=       1 Green=       1",4,""
LogMemory,       2023-11-09T03:34:32,      622352,"CycleColors                   ","Count: Red=       1 Blue=       0 Green=       1",2,""
```
...
```
LogMemory,       2023-11-09T07:13:00,      681024,"CycleColors                   ","Count: Red=    8350 Blue=    8351 Green=    8350",37299,""
LogMemory,       2023-11-09T07:13:00,      681232,"CycleColors                   ","Count: Red=    8350 Blue=    8351 Green=    8351",37300,""
LogMemory,       2023-11-09T07:13:00,      681440,"CycleColors                   ","Count: Red=    8351 Blue=    8351 Green=    8351",37301,""
```

alexischr, 2023-11-10:
The slowly raising GC memory profile should not be a problem if it eventually triggers a GC collection, which may not happen until more memory pressure than 700kb is exercised. So, your latest run is looking good @doingnz !

doingnz, 2023-11-10:
The code that dumps the memory used forces a GC before every Log line.

closed 2024-03-05 by alexischr

doingnz, 2024-03-06:
It would be good to clarify the original issue raised here related to `TaskCancelledException` is fixed.

The memory leak when the Debugger is attached is not fixed, but is being tracked in other issues. 
For example [#387](https://github.com/WildernessLabs/Meadow_Issues/issues/387).