WildernessLabs/Meadow.Foundation #753: MCP23017 and MCP23S17 throw null reference when accessing pins

asieradzk, 2023-08-29:
**Describe the bug**
From meadow foundation:
MCP23017 and MCP23S17 both throw null exception when trying to access pins after they had been instantiated/initialized

I checked that MCP23008 works fine. 
Changing the definition of MCP from 17 to 08 (While 17 is hooked up) makes code compile correctly (pins of course dont work - as expected). This simple experiment tells me there is something wrong with how 17 and S17 works under the hood

Also tried two different boards - both same result. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ''http://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.ICs.IOExpanders.Mcp23017.html"
2. Copy code
3. Hook up MCP23017
4. See error: Null reference

**Expected behavior**
No null reference - code should not crash, should be possible to access read/write from pins

**Developer tools (please complete the following information as best as you can):**
 - WIndows 11
 - Visual Studio 2022
 - Meadow OS/Mono/coprocessor/Nuget: Tried 1.0.2.1 and 1.1 (in sync and various permutations)
 - F7v2

Output:
```
...
Device is configured to use WiFi for the network interface
Update Service is disabled.
App Error in App Run: Object reference not set to an instance of an object.
System.NullReferenceException: Object reference not set to an instance of an object.
at MeadowApplication1.MeadowApp.TestDigitalOutputPorts (System.Int32 loopCount) [0x00012] in <8718cd4b0124436db3920125374e8a0e>:0
at MeadowApplication1.MeadowApp.Run () [0x0000d] in <8718cd4b0124436db3920125374e8a0e>:0
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x00111] in <3099356e20a447dfb5e0917f6484795c>:0
```

adrianstevens, 2023-09-06:
I'm curious what's null in TestDigitalOutputPorts. I was able to run the sample app in Meadow.Foundation here:
https://github.com/WildernessLabs/Meadow.Foundation/tree/develop/Source/Meadow.Foundation.Peripherals/ICs.IOExpanders.Mcp23xxx/Samples/Mcp23017_Sample 

Can you debug or add some Console.Writeline calls to check if the mcp instance not-null?

adrianstevens, 2023-09-14:
Can't repo so I'm closing the issue but please feel free to re-open with further details

closed 2023-09-14 by adrianstevens