WildernessLabs/Meadow.Foundation #763: microLayout is unstable when clearing the scree

bryancostanich, 2023-09-08:
**Describe the bug**
I've been trying to use microLayout to do multiple screens. the first test was to display a splash screen, clear it, and then display a menu. but when i clear the controls, the app crashes. well, it doesn't crash directly, but when i `return` out of `App.Run()`, it crashes. 

**To Reproduce**
Run the attached [RelayControl.zip](https://github.com/WildernessLabs/Meadow.Foundation/files/12563577/RelayControl.zip) app and uncomment out the commented lines in `Run()`:

```csharp
public override Task Run()
{
    //SplashScreen.Show(screen);

    //Task.Delay(1000);

    //Resolver.Log.Info("splash screen going down.");

    //screen.Controls.Clear();

    //Resolver.Log.Info("cleared screen.");

    ShowMenuScreen();

    return base.Run();
}
```

**Expected behavior**
Screen should clear, and menu should show.

**Meadow.Foundation (please complete the following information as best as you can):**
 - Meadow.Foundation v1.3.0

**Developer tools (please complete the following information as best as you can):**
 - OS and version: macOS ventura 13.4.1
 - IDE and version: VS4Mac 17.6.3 (build 421)
 - v1.3.0
 
**Meadow (please complete the following information as best as you can):**
 - Meadow hardware version: Project Lab V3
 - Meadow OS version: 1.3.0

patridge, 2023-09-08:
In this case, I think there is a split of problems. There seems to be an issue when things crash here that causes VS to get angry. But in the sample itself, not using `async`/`await` is causing the code to execute in weird ways.

## Handling async in the sample

I was able to get the app to run without an error by swapping for this awaited code.

```csharp
public override async Task Run()
{
    SplashScreen.Show(screen);

    await Task.Delay(1000);

    Resolver.Log.Info("splash screen going down.");

    screen.Controls.Clear();

    Resolver.Log.Info("cleared screen.");

    //ShowMenuScreen();

    await base.Run();
}
```

## VS issues when an app blows up

When I debug on Windows VS2022, I get several symptoms of problems handling errors.

[image]
[image]
[image]

ctacke, 2023-09-08:
The controls collection was not synchronized and the lack of an await on your Task.Delay uncovered it, so your bug uncovered an actual bug, huzzah!  Fixed in the LineChart branch, so when the PR is applied, it will fix this issue as well.

closed 2023-09-26 by adrianstevens