WildernessLabs/Meadow.Foundation #950: Relay activates on creation

youtoofan, 2024-04-17:
I have an instance of a relay, using a Pin and RelayType.NormallyOpen
During boot, everything seems normal.

In the Initialization method, I create the relay instance and the relay switches state and thus activates.
I can hear (click) and see (led) it was activated.
If i press a button, i can toggle the relay state, but it is inversed.

**Reproduce**
1. Connect the relay to power and a meadow port (i tried with ports 5 and 6)
2. Create an instance of the relay in the Initialize method
3. Put a breakpoint before the init
4. Debug
5. VS stops at the breakpoint
6. Relay is inactive
7. Step over until after relay creation
8. Relay activates

`
// inactive

var temp = new Relay(Device.Pins.D06);

// click and active
`

**Expected behavior**
The relay should stay open after initialization. (Or closed if you have the normally closed version)

This is the type of relay i used.
[image]

**Meadow.Foundation (please complete the following information as best as you can):**
 - NuGet package version: Meadow.Foundation.Relays 1.10.2

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11 
 - IDE and version: Visual Studio Community 2022 
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.9.7

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7v2
 - Meadow OS version: 1.10.0.2

**Additional context**
I noticed this behaviour in some older versions too probably, but only now i have a piece of hardware connected to the relay so it became apparent.

youtoofan, 2024-04-17:
Seems the issue is not really related to the relay, but is cause by the DigitalOutputPort

`  
pin.CreateDigitalOutputPort(false);
`

After creating the outputPort, the relay activates.

ctacke, 2024-04-17:
I'd need to see a schematic on how it's wired.

The constructor you use defaults to `NormallyOpen`:
https://github.com/WildernessLabs/Meadow.Foundation/blob/cc60d28c92a96b357b5d3539f9312fd92d4ed7cc/Source/Meadow.Foundation.Core/Relays/Relay.cs#L52

It creates a `DigitalOutputPort` that defaults to being low, so the `D06` should be low/pulled to ground.  Depending on how the relay is wired, this might activate or deactivate the coil.

youtoofan, 2024-04-18:
This is the best i could do, using an arduino pico

[image]

I measure continuity with a multimeter between the normally open side of the relay and the common pin when the DigitalOutputPort is created. I can hear it click and the built-in led lights up. (Which is not ok)

youtoofan, 2024-04-18:
Seems for this particular relay, I need to set the initial state of the digital output port to true.
Then the relay stays open until i activate it. Issue can be closed.

closed 2024-04-18 by youtoofan

ctacke, 2024-04-18:
I suspect it's internally (on the PCB) wired with negative logic.